swiftlang/swift-embedded-examples #4: `stm32-uart-echo` Build Error

fumoboy007, 2024-04-05:
I am trying to build the` stm32-uart-echo` example but not sure what I am doing wrong. I get the following error:
```
<unknown>:0: error: unable to load standard library for target 'armv7em-apple-none-macho'
```

My build command:
```sh
TOOLCHAINS="Swift Development Snapshot 2024-04-03 (a)" make
```

Any ideas?

MaxDesiatov, 2024-04-06:
What's the output of `swift --version` after setting `TOOLCHAINS` this way? I suspect the value of `TOOLCHAINS` is wrong, I've never seen it set in this format, it usually expects a plist key instead of a human readable version string.

fumoboy007, 2024-04-06:
Hi, @MaxDesiatov! Both `Swift Development Snapshot 2024-04-03 (a)` and `org.swift.59202404031a` produce the same result:
```
% TOOLCHAINS="Swift Development Snapshot 2024-04-03 (a)" swift --version
Apple Swift version 6.0-dev (LLVM cb4f50b342f3cf9, Swift fc5da5342ce965c)
Target: arm64-apple-macosx14.0
```

Here is the full build log (with some newlines to improve readability):
```
% TOOLCHAINS=org.swift.59202404031a make   
           
compling...
/Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/bin/swift-build \
		--configuration release \
		--verbose \
		--triple armv7em-apple-none-macho \
		-Xcc -D__APPLE__ -Xcc -D__MACH__ \
		-Xswiftc -Xfrontend -Xswiftc -disable-stack-protector

warning: 'stm32-uart-echo': /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/bin/swift-frontend -frontend -c -primary-file /Users/fumoboy007/Downloads/swift-embedded-examples/stm32-uart-echo/Package.swift -target arm64-apple-macosx13.0 -Xllvm -aarch64-use-tbi -enable-objc-interop -sdk /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.4.sdk -I /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/usr/lib -I /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/lib/swift/pm/ManifestAPI -F /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/Library/Frameworks -vfsoverlay /var/folders/kn/9dnh9mbx24zcltl6p8syx1w00000gn/T/TemporaryDirectory.fIYEGn/vfs.yaml -swift-version 5 -package-description-version 5.10.0 -disable-implicit-concurrency-module-import -disable-implicit-string-processing-module-import -empty-abi-descriptor -Xcc -working-directory -Xcc /Users/fumoboy007/Downloads/swift-embedded-examples/stm32-uart-echo -resource-dir /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/lib/swift -module-name main -target-sdk-version 14.4 -target-sdk-name macosx14.4 -external-plugin-path '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/usr/lib/swift/host/plugins#/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/usr/bin/swift-plugin-server' -external-plugin-path '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/usr/local/lib/swift/host/plugins#/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/usr/bin/swift-plugin-server' -plugin-path /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/lib/swift/host/plugins -plugin-path /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/local/lib/swift/host/plugins -o /var/folders/kn/9dnh9mbx24zcltl6p8syx1w00000gn/T/TemporaryDirectory.oSAmSg/Package-1.o

/Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/bin/clang /var/folders/kn/9dnh9mbx24zcltl6p8syx1w00000gn/T/TemporaryDirectory.oSAmSg/Package-1.o -F /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/Library/Frameworks --sysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.4.sdk --target=arm64-apple-macosx13.0 /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/lib/swift/macosx/libswiftCompatibilityPacks.a -L /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/lib/swift/macosx -L /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.4.sdk/usr/lib/swift -L /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/lib/swift/pm/ManifestAPI -L /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/usr/lib -L /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/lib/swift/pm/ManifestAPI -L /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/usr/lib -lPackageDescription -Xlinker -rpath -Xlinker /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/lib/swift/pm/ManifestAPI -o /var/folders/kn/9dnh9mbx24zcltl6p8syx1w00000gn/T/TemporaryDirectory.E0czcf/stm32-uart-echo-manifest

Apple Swift version 6.0-dev (LLVM cb4f50b342f3cf9, Swift fc5da5342ce965c)
Target: arm64-apple-macosx13.0
Planning build

warning: 'swift-mmio': /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/bin/swift-frontend -frontend -c -primary-file /Users/fumoboy007/Downloads/swift-embedded-examples/stm32-uart-echo/.build/checkouts/swift-mmio/Package.swift -target arm64-apple-macosx13.0 -Xllvm -aarch64-use-tbi -enable-objc-interop -sdk /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.4.sdk -I /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/usr/lib -I /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-03-a.xctoolchain/usr/lib/swift/pm/ManifestAPI -F /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/Library/Frameworks -vfsoverlay /var/folders/kn/9dnh9mbx24zcltl6p8syx1w00000gn/T/TemporaryDirectory.sn4R1Z/vfs.yaml -swift-version 5 -package-description-version 5.9.0 -disable-implicit-concurrency-module-import -disable-implicit-string-processing-module-import -empty-abi-descriptor -Xcc -working-directory -Xcc /Users/fumoboy007/Downloads/swift-embedded-examples/stm32-uart-echo [... cut ...] /Downloads/swift-embedded-examples/stm32-uart-echo/.build/armv7em-apple-none-macho/release/MMIOUtilities.build/Parser.swift.o -o /Users/fumoboy007/Downloads/swift-embedded-examples/stm32-uart-echo/.build/armv7em-apple-none-macho/release/MMIOUtilities.build/StringInterpolation+Hex.swift.o

<unknown>:0: error: unable to load standard library for target 'armv7em-apple-none-macho'
<unknown>:0: error: unable to load standard library for target 'armv7em-apple-none-macho'

make: *** [build] Error 1
```

fumoboy007, 2024-04-06:
There seem to be a lot of references to Xcode. I wonder if that’s expected.

unnamedd, 2024-04-06:
@fumoboy007 give a try checking the [swift-playdate-examples](https://github.com/apple/swift-playdate-examples) source code, especially [this part here](https://github.com/apple/swift-playdate-examples/blob/main/Examples/swift.mk#L21-L29) where the toolchain is being taken automatically to help in the compilation.

fumoboy007, 2024-04-09:
@unnamedd As I have shown above, I think the correct toolchain is being selected. Thanks for the additional example though!

andelf, 2024-04-14:
Same here. Any ideas?
`-enable-experimental-feature Embedded` is not added while building, this might be the reason why standard library is required.
If I added this, then swift-mmio dependency fails complaining about `String` is not available in embedded Swift.

kubamracek, 2024-07-08:
I think (a part of) the problem is the `TOOLCHAINS="Swift Development Snapshot 2024-07-07 (a)"` syntax -- this doesn't actually select the right toolchain. Use the `TOOLCHAINS=org.swift.59202407071a` syntax.

I have just tried building the `stm32-uart-echo` example with the latest downloadable swift.org toolchain and it builds correctly. @fumoboy007 can you try again, too?

fumoboy007, 2024-07-14:
It’s not related to the syntax. It seems something was broken in the 2024-04-03 Swift toolchain. 2024-07-11 works now.

closed 2024-07-14 by fumoboy007