swiftlang/swift-embedded-examples #5: `pico-blink` Build Error

star-light-1, 2024-04-18:
When trying to build the `pico-blink` example on macOS with:

```
cd swift-embedded-examples/pico-blink
rm -rf .build ; TOOLCHAINS=org.swift.59202404131a ./build.sh
```

I get the following error:

```
# Link
$CLANG .build/release/Support.build/{Support.c,crt0.S}.o .build/release/Blinky.build/*.o -target armv6m-apple-none-macho -o $BUILDROOT/blinky $LD_FLAGS
+ /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-13-a.xctoolchain/usr/bin/clang .build/release/Support.build/Support.c.o .build/release/Support.build/crt0.S.o .build/release/Blinky.build/Blinky.swift.o -target armv6m-apple-none-macho -o /path/to/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky -static -Wl,-e,_reset -dead_strip -Wl,-no_zero_fill_sections -Wl,-segalign,4 -Wl,-segaddr,__RESET,0x20000000 -Wl,-segaddr,__VECTORS,0x20000100 -Wl,-seg1addr,0x20000200 -Wl,-pagezero_size,0 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/15.0.0/lib/darwin/macho_embedded/libclang_rt.soft_static.a
ld: warning: ignoring file /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-13-a.xctoolchain/usr/lib/clang/17/lib/darwin/macho_embedded/libclang_rt.soft_static.a, building for free standing-armv6m but attempting to link with file built for macOS-armv7
ld: warning: ignoring file /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/15.0.0/lib/darwin/macho_embedded/libclang_rt.soft_static.a, building for free standing-armv6m but attempting to link with file built for macOS-armv7
Undefined symbols for architecture armv6m:
  "___muldi3", referenced from:
      _$s6Blinky4MainV5delayyySiFZTf4nd_n in Blinky.swift.o
ld: symbol(s) not found for architecture armv6m
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Is the `___muldi3` symbol not available on armv6m? If not, why is it being used?

System information:

```
swift-DEVELOPMENT-SNAPSHOT-2024-04-13-a.xctoolchain
Apple Swift version 6.0-dev (LLVM 01231f56c62f4c4, Swift abaa63821080b92)
Darwin Kernel Version 23.4.0: Fri Mar 15 00:12:25 PDT 2024; root:xnu-10063.101.17~1
```

Toolchain ID:
```
$ plutil -extract CFBundleIdentifier raw -o - /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-*.xctoolchain/Info.plist                                    
org.swift.59202404131a
```

Git commit of `swift-embedded-examples`: 
```
69dc23e718c5b20797daec17d4754cd42bb21c24
```

Full build log: [pico-blink-build-log.txt](https://github.com/apple/swift-embedded-examples/files/15029563/pico-blink-build-log.txt)

Joannis, 2024-04-26:
@star-light-1 `___muldi3` multiplies two 64-bit integers. The `100 * N` is where multiplication is used in `func delay(_ N: Int)`

Joannis, 2024-04-26:
Try using the toolchain from April 3rd, that worked for me.

star-light-1, 2024-04-26:
The oldest toolchain I could find in section "Trunk Development (main)" on [swift.org/download](https://www.swift.org/download) is [April 4, 2024](https://download.swift.org/development/xcode/swift-DEVELOPMENT-SNAPSHOT-2024-04-04-a/swift-DEVELOPMENT-SNAPSHOT-2024-04-04-a-osx.pkg), which for me produces the same error shown above.
Latest toolchain of the [5.10 branch](https://download.swift.org/swift-5.10-branch/xcode/swift-5.10-DEVELOPMENT-SNAPSHOT-2024-03-02-a/swift-5.10-DEVELOPMENT-SNAPSHOT-2024-03-02-a-osx.pkg) produces:
```
error: unable to load standard library for target 'armv6m-apple-none-macho'
```

kubamracek, 2024-04-27:
I can sort-of reproduce the problem. One issue is that this part of the build script for pico-blink should not be necessary anymore, as the expectation is that the downloadable swift.org toolchain includes a valid libclang_rt.soft_static.a for the CPU arch we're building here (armv6m):

```
XCODE_RESOURCE_DIR=`env TOOLCHAINS="" xcrun clang -print-resource-dir`
LD_FLAGS+=" $XCODE_RESOURCE_DIR/lib/darwin/macho_embedded/libclang_rt.soft_static.a"
```

Once I drop this, I get:

```
# Link
$CLANG .build/release/Support.build/{Support.c,crt0.S}.o .build/release/Blinky.build/*.o -target armv6m-apple-none-macho -o $BUILDROOT/blinky $LD_FLAGS
+ /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-25-a.xctoolchain/usr/bin/clang .build/release/Support.build/Support.c.o .build/release/Support.build/crt0.S.o .build/release/Blinky.build/Blinky.swift.o -target armv6m-apple-none-macho -o /Users/kuba/Documents/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky -static -Wl,-e,_reset -dead_strip -Wl,-no_zero_fill_sections -Wl,-segalign,4 -Wl,-segaddr,__RESET,0x20000000 -Wl,-segaddr,__VECTORS,0x20000100 -Wl,-seg1addr,0x20000200 -Wl,-pagezero_size,0
ld: warning: ignoring file /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-04-25-a.xctoolchain/usr/lib/clang/17/lib/darwin/macho_embedded/libclang_rt.soft_static.a, building for free standing-armv6m but attempting to link with file built for macOS-armv7
Undefined symbols for architecture armv6m:
  "___muldi3", referenced from:
      _$s6Blinky4MainV5delayyySiFZTf4nd_n in Blinky.swift.o
ld: symbol(s) not found for architecture armv6m
```

The problem is that libclang_rt.soft_static.a is somehow only armv7 and contains no other slices. That will need to be fixed, but I'm not sure how that happens in the first place.

The only workaround I can think of (short of actually removing the multiplication from source code, but sometimes those are really useful 😄) would be to rebuild libclang_rt from LLVM's compiler-rt from source and explicitly build the armv6m slice, and link that instead.

kubamracek, 2024-04-27:
@star-light-1 can I also ask which Xcode version are you using?

star-light-1, 2024-04-29:
This happend using Xcode 15.3 (15E204a)

Bauer312, 2024-05-05:
The most recent 5.10 (May 1, 2024) and 6.0 (April 30, 2024) snapshot both still produce the linker warning
`ld: warning: ignoring file /<SNIP>/macho_embedded/libclang_rt.soft_static.a, building for free standing-armv6m but attempting to link with file built for macOS-armv7`

5.10 fails with the undefined symbol, while 6.0 continues and produces the expected blinky.uf2 output, although not in the location that the README says it will appear.  I have a pico w and it appears to reject loading it, but I need to do more testing to see if this is unrelated.

TG-Techie, 2024-06-17:
I have a (admittedly poorly maintainable) patch that works for adding armv6m `___muldi3`

https://gist.github.com/TG-Techie/5be0cff9457bce50752c69eced81864f

Or....
>  removing the multiplication from source code, but sometimes those are really useful 😄

TG-Techie, 2024-06-18:
update /  re: @kubamracek 's suggestion of re-buliding a slice for armv6m, 

Here is a version of the clang_rt I build from the main trunk of the llvm-project (753498e should be).

### Usage 
 I copied the `.a` into the the `Sources/Support` directory and then added it as the first argument on this line
https://github.com/apple/swift-embedded-examples/blob/2b93644fc4cca46c48e8da3023746c8d9e26007e/pico-blink/build.sh#L38

It worked well for the blink test so far 💁‍♂️

[libclang_rt.soft_static_armv6m_macho_embedded.a.zip](https://github.com/user-attachments/files/15878757/libclang_rt.soft_static_armv6m_macho_embedded.a.zip)

Happy Hacking,
    ~ TG-Techie

niqt, 2024-06-24:
I added what @TG-Techie said, but now i have this error:
# Link
$CLANG Sources/Support/libclang_rt.soft_static_armv6m_macho_embedded.a .build/release/Support.build/{Support.c,crt0.S}.o .build/release/Blinky.build/*.o -target armv6m-apple-none-macho -o $BUILDROOT/blinky $LD_FLAGS
+ /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-06-13-a.xctoolchain/usr/bin/clang Sources/Support/libclang_rt.soft_static_armv6m_macho_embedded.a .build/release/Support.build/Support.c.o .build/release/Support.build/crt0.S.o .build/release/Blinky.build/Blinky.swift.o -target armv6m-apple-none-macho -o /Users/nicola/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky -static -Wl,-e,_reset -dead_strip -Wl,-no_zero_fill_sections -Wl,-segalign,4 -Wl,-segaddr,__RESET,0x20000000 -Wl,-segaddr,__VECTORS,0x20000100 -Wl,-seg1addr,0x20000200 -Wl,-pagezero_size,0 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/15.0.0/lib/darwin/macho_embedded/libclang_rt.soft_static.a
ld: warning: ignoring file /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/15.0.0/lib/darwin/macho_embedded/libclang_rt.soft_static.a, building for free standing-armv6m but attempting to link with file built for macOS-armv7
ld: warning: ignoring file /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-06-13-a.xctoolchain/usr/lib/clang/17/lib/darwin/macho_embedded/libclang_rt.soft_static.a, building for free standing-armv6m but attempting to link with file built for macOS-armv7

# Extract sections from executable into flashable binary
$PYTHON_EXEC $MACHO2UF2 $BUILDROOT/blinky $BUILDROOT/blinky.uf2 --base-address 0x20000000 --segments '__TEXT,__DATA,__VECTORS,__RESET'
+ /Applications/Xcode.app/Contents/Developer/usr/bin/python3 /Users/nicola/swift-embedded-examples/Tools/macho2uf2.py /Users/nicola/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky /Users/nicola/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky.uf2 --base-address 0x20000000 --segments __TEXT,__DATA,__VECTORS,__RESET
Traceback (most recent call last):
  File "/Users/nicola/swift-embedded-examples/Tools/macho2bin.py", line 32, in <module>
    from macholib import MachO
ModuleNotFoundError: No module named 'macholib'
Traceback (most recent call last):
  File "/Users/nicola/swift-embedded-examples/Tools/macho2uf2.py", line 89, in <module>
    main()
  File "/Users/nicola/swift-embedded-examples/Tools/macho2uf2.py", line 47, in main
    subprocess.check_call([MY_DIR + "/macho2bin.py", args.input, args.input + ".bin", "--base-address", "0x%08x" % args.base_address, "--segments", ",".join(args.segments)])
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/Users/nicola/swift-embedded-examples/Tools/macho2bin.py', '/Users/nicola/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky', '/Users/nicola/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky.bin', '--base-address', '0x20000000', '--segments', '__TEXT,__DATA,__VECTORS,__RESET']' returned non-zero exit status 1.

TG-Techie, 2024-06-25:
@niqt , three follow ups:
a) What version of the toolchain are you running? 
b) Would you mind adding ` ``` ` ticks to your above post for readability, please? 
c) it looks like you may need to install the python requirements.txt in the repo's tool folder based on these lines in the above text:
>  ```
>  from macholib import MachO
>  ModuleNotFoundError: No module named 'macholib'
>  ```

niqt, 2024-06-25:
> 
> > ```
> > from macholib import MachO
> > ModuleNotFoundError: No module named 'macholib'
> > ```

Yeah, it was missed python dependencies...

kubamracek, 2024-07-09:
I'll use this issue to track fixing the `Undefined symbols for architecture armv6m: "___muldi3"` problem, for other issues please file a separate problem report.

closed 2024-07-23 by kubamracek

kubamracek, 2024-07-30:
Just tried the latest downloadable swift.org toolchain (swift-DEVELOPMENT-SNAPSHOT-2024-07-29-a.xctoolchain) and I no longer see this problem. I think this actually is finally resolved.