swiftlang/swift-embedded-examples #7: `pico-blink-sdk` can't find Pico W headers

MaxDesiatov, 2024-04-28:
When I add `#include "pico/cyw43_arch.h"` (header used in `picow_blink.c` in upstream `pico-examples`) to `BridgingHeader.h` and reconfigure with `cmake -DPICO_BOARD=pico_w -B build -G Ninja .` that runs fine, I see this output from CMake:

```
PICO platform is rp2040.
Build type is Release
Using PICO_BOARD from environment ('pico_w')
Using CMake board configuration from /Users/maxd/Documents/embedded/pico-sdk/src/boards/pico_w.cmake
Using board configuration from /Users/maxd/Documents/embedded/pico-sdk/src/boards/include/boards/pico_w.h
TinyUSB available at /Users/maxd/Documents/embedded/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /Users/maxd/Documents/embedded/pico-sdk/lib/btstack
cyw43-driver available at /Users/maxd/Documents/embedded/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /Users/maxd/Documents/embedded/pico-sdk/lib/lwip
Pico W Wi-Fi build support available.
```

But when I try to build with `cmake --build build` it can't find any of the headers specific to Pico W:

```
BridgingHeader.h:15:10: error: 'pico/cyw43_arch.h' file not found
13 | 
14 | #include "pico/stdlib.h"
15 | #include "pico/cyw43_arch.h"
   |          `- error: 'pico/cyw43_arch.h' file not found
```

I tried adding include paths manually:

```
add_custom_command(
    OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
    COMMAND
        ${SWIFTC}
        -target armv6m-none-none-eabi -Xcc -mfloat-abi=soft -Xcc -fshort-enums
        -Xfrontend -function-sections -enable-experimental-feature Embedded -wmo -parse-as-library
        $$\( echo '$<TARGET_PROPERTY:swift-blinky,INCLUDE_DIRECTORIES>' | tr '\;' '\\n' | sed -e 's/\\\(.*\\\)/-Xcc -I\\1/g' \)
        $$\( echo '${CMAKE_C_IMPLICIT_INCLUDE_DIRECTORIES}'             | tr ' '  '\\n' | sed -e 's/\\\(.*\\\)/-Xcc -I\\1/g' \)
        -I$ENV{PICO_SDK_PATH}/src/rp2_common/pico_cyw43_arch/include
        -I$ENV{PICO_SDK_PATH}/src/rp2_common/pico_cyw43_driver/include
        -I$ENV{PICO_SDK_PATH}/src/rp2_common/pico_async_context/include/
        -I$ENV{PICO_SDK_PATH}/lib/cyw43-driver/src
        -I$ENV{PICO_SDK_PATH}/lib/lwip/src/include
        -I$ENV{PICO_SDK_PATH}/lib/lwip/contrib/ports/unix/lib
        -I$ENV{PICO_SDK_PATH}/lib/lwip/contrib/ports/unix/port/include
        -import-bridging-header ${CMAKE_CURRENT_LIST_DIR}/BridgingHeader.h
        ${CMAKE_CURRENT_LIST_DIR}/Main.swift
        -c -o ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
    DEPENDS
        ${CMAKE_CURRENT_LIST_DIR}/BridgingHeader.h
        ${CMAKE_CURRENT_LIST_DIR}/Main.swift
)
```

but this seems wrong, as `picow_blink` from `pico-examples` doesn't seem to be doing this and gets its include headers automatically.

It also doesn't help anyway as in the end I get this error:

```
pico-sdk/src/rp2_common/pico_cyw43_arch/include/pico/cyw43_arch.h:30:2: error: must specify support pico_cyw43_arch architecture type or set PICO_CYW43_ARCH_HEADER
 28 | #include "pico/cyw43_arch/arch_freertos.h"
 29 | #else
 30 | #error must specify support pico_cyw43_arch architecture type or set PICO_CYW43_ARCH_HEADER
    |  `- error: must specify support pico_cyw43_arch architecture type or set PICO_CYW43_ARCH_HEADER
 31 | #endif
 32 | #endif
```

MaxDesiatov, 2024-04-28:
When working with code not specific to Pico W, this also happens to headers like `"hardware/i2c.h"`

willswire, 2024-06-19:
Have you verified that your `target_link_libraries` include the proper libs?

```
# Link the Swift object file with the executable target
target_link_libraries(swift-pico
    pico_stdlib
    hardware_uart
    hardware_gpio
    pico_lwip_arch
    pico_cyw43_arch_none
    ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
)
```

shivarajd, 2024-07-06:
Have you set the PICO_SDK_PATH environment variable? 

I have created a Docker container (https://github.com/shivarajd/swift-embedded-pico-docker) and everything works fine.

kubamracek, 2024-07-09:
As contributed by @yochidros in https://github.com/apple/swift-embedded-examples/pull/11, we need a slightly different setup for Pico W compared to Pico. Making this even more explicit in the docs in https://github.com/apple/swift-embedded-examples/pull/31. With that I think this should be resolved.

closed 2024-07-09 by kubamracek