swiftlang/swift-embedded-examples #12: pico's led doesn't blink

ilgrandeanonimo, 2024-05-14:
Today I tried to run the Blinky example on my Raspberry Pico Pi W.
I sucefully builded the .uf2 file:
- Installed Swift 6.0-dev Nightly
- Cloned Repo
- cd swift-embedded-examples/pico-blink
- Changed the Toolchain:
```sh
export TOOLCHAINS=$(plutil -extract CFBundleIdentifier raw \ 
~/Library/Developer/Toolchains/swift-6.0-DEVELOPMENT-SNAPSHOT-2024-04-30-a.xctoolchain/Info.plist)
```
- compiled:
```sh
./build
```

but when i install the .uf2 file the led doesn't blink

Build Output:
```
# Determine file paths
REPOROOT=$(git rev-parse --show-toplevel)
git rev-parse --show-toplevel
++ git rev-parse --show-toplevel
+ REPOROOT=/Users/ilgrandeanonimo/swift-embedded-examples
TOOLSROOT=$REPOROOT/Tools
+ TOOLSROOT=/Users/ilgrandeanonimo/swift-embedded-examples/Tools

# Setup tools and build flags
SWIFT_EXEC=${SWIFT_EXEC:-`xcrun -f swift`}
xcrun -f swift
++ xcrun -f swift
+ SWIFT_EXEC=/Users/ilgrandeanonimo/Library/Developer/Toolchains/swift-6.0-DEVELOPMENT-SNAPSHOT-2024-04-30-a.xctoolchain/usr/bin/swift
CLANG=${CLANG:-`xcrun -f clang`}
xcrun -f clang
++ xcrun -f clang
+ CLANG=/Users/ilgrandeanonimo/Library/Developer/Toolchains/swift-6.0-DEVELOPMENT-SNAPSHOT-2024-04-30-a.xctoolchain/usr/bin/clang
SWIFT_FLAGS="-enable-experimental-feature Embedded -disable-stack-protector"
+ SWIFT_FLAGS='-enable-experimental-feature Embedded -disable-stack-protector'
CLANG_FLAGS="-D__MACH__ -ffreestanding -mcpu=cortex-m0plus -mthumb"
+ CLANG_FLAGS='-D__MACH__ -ffreestanding -mcpu=cortex-m0plus -mthumb'
LD_FLAGS="-static -Wl,-e,_reset -dead_strip -Wl,-no_zero_fill_sections -Wl,-segalign,4 -Wl,-segaddr,__RESET,0x20000000 -Wl,-segaddr,__VECTORS,0x20000100 -Wl,-seg1addr,0x20000200 -Wl,-pagezero_size,0"
+ LD_FLAGS='-static -Wl,-e,_reset -dead_strip -Wl,-no_zero_fill_sections -Wl,-segalign,4 -Wl,-segaddr,__RESET,0x20000000 -Wl,-segaddr,__VECTORS,0x20000100 -Wl,-seg1addr,0x20000200 -Wl,-pagezero_size,0'

XCODE_RESOURCE_DIR=`env TOOLCHAINS="" xcrun clang -print-resource-dir`
env TOOLCHAINS="" xcrun clang -print-resource-dir
++ env TOOLCHAINS= xcrun clang -print-resource-dir
+ XCODE_RESOURCE_DIR=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/15.0.0
LD_FLAGS+=" $XCODE_RESOURCE_DIR/lib/darwin/macho_embedded/libclang_rt.soft_static.a"
+ LD_FLAGS+=' /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/15.0.0/lib/darwin/macho_embedded/libclang_rt.soft_static.a'

SWIFT_BUILD_FLAGS="--triple armv6m-apple-none-macho --configuration release --verbose"
+ SWIFT_BUILD_FLAGS='--triple armv6m-apple-none-macho --configuration release --verbose'
for SWIFT_FLAG in $SWIFT_FLAGS; do
	SWIFT_BUILD_FLAGS+=" -Xswiftc $SWIFT_FLAG"
done
+ for SWIFT_FLAG in '$SWIFT_FLAGS'
+ SWIFT_BUILD_FLAGS+=' -Xswiftc -enable-experimental-feature'
+ for SWIFT_FLAG in '$SWIFT_FLAGS'
+ SWIFT_BUILD_FLAGS+=' -Xswiftc Embedded'
+ for SWIFT_FLAG in '$SWIFT_FLAGS'
+ SWIFT_BUILD_FLAGS+=' -Xswiftc -disable-stack-protector'

for CLANG_FLAG in $CLANG_FLAGS; do
	SWIFT_BUILD_FLAGS+=" -Xcc $CLANG_FLAG"
done
+ for CLANG_FLAG in '$CLANG_FLAGS'
+ SWIFT_BUILD_FLAGS+=' -Xcc -D__MACH__'
+ for CLANG_FLAG in '$CLANG_FLAGS'
+ SWIFT_BUILD_FLAGS+=' -Xcc -ffreestanding'
+ for CLANG_FLAG in '$CLANG_FLAGS'
[... 64 lines cut ...]

# Link
$CLANG .build/release/Support.build/{Support.c,crt0.S}.o .build/release/Blinky.build/*.o -target armv6m-apple-none-macho -o $BUILDROOT/blinky $LD_FLAGS
+ /Users/ilgrandeanonimo/Library/Developer/Toolchains/swift-6.0-DEVELOPMENT-SNAPSHOT-2024-04-30-a.xctoolchain/usr/bin/clang .build/release/Support.build/Support.c.o .build/release/Support.build/crt0.S.o .build/release/Blinky.build/Blinky.swift.o -target armv6m-apple-none-macho -o /Users/ilgrandeanonimo/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky -static -Wl,-e,_reset -dead_strip -Wl,-no_zero_fill_sections -Wl,-segalign,4 -Wl,-segaddr,__RESET,0x20000000 -Wl,-segaddr,__VECTORS,0x20000100 -Wl,-seg1addr,0x20000200 -Wl,-pagezero_size,0 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/15.0.0/lib/darwin/macho_embedded/libclang_rt.soft_static.a
ld: warning: ignoring file /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/15.0.0/lib/darwin/macho_embedded/libclang_rt.soft_static.a, building for free standing-armv6m but attempting to link with file built for macOS-armv7
ld: warning: ignoring file /Users/ilgrandeanonimo/Library/Developer/Toolchains/swift-6.0-DEVELOPMENT-SNAPSHOT-2024-04-30-a.xctoolchain/usr/lib/clang/17/lib/darwin/macho_embedded/libclang_rt.soft_static.a, building for free standing-armv6m but attempting to link with file built for macOS-armv7

# Extract sections from executable into flashable binary
$PYTHON_EXEC $MACHO2UF2 $BUILDROOT/blinky $BUILDROOT/blinky.uf2 --base-address 0x20000000 --segments '__TEXT,__DATA,__VECTORS,__RESET'
+ /Applications/Xcode.app/Contents/Developer/usr/bin/python3 /Users/ilgrandeanonimo/swift-embedded-examples/Tools/macho2uf2.py /Users/ilgrandeanonimo/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky /Users/ilgrandeanonimo/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky.uf2 --base-address 0x20000000 --segments __TEXT,__DATA,__VECTORS,__RESET
Writing segment __RESET size 24 (0x24) at vmaddr 0x20000000
Writing gap of size 232 (0xe8) at vmaddr 0x20000018
Writing segment __VECTORS size 192 (0x192) at vmaddr 0x20000100
Writing gap of size 64 (0x40) at vmaddr 0x200001c0
Writing segment __TEXT size 4940 (0x4940) at vmaddr 0x20000200
Writing gap of size 20 (0x14) at vmaddr 0x2000154c
Writing segment __DATA size 2064 (0x2064) at vmaddr 0x20001560
Produced /Users/ilgrandeanonimo/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky.bin with 7536 bytes
Produced /Users/ilgrandeanonimo/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky.uf2 with total size 15360 (0x3c00) bytes, payload size 7536 (0x1d70) bytes

# Echo final binary path
ls -al $BUILDROOT/blinky.uf2
+ ls -al /Users/ilgrandeanonimo/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky.uf2
-rw-r--r--@ 1 ilgrandeanonimo  staff  15360 14 Mag 14:08 /Users/ilgrandeanonimo/swift-embedded-examples/pico-blink/.build/armv6m-apple-none-macho/release/blinky.uf2
```

phausler, 2024-06-13:
The pico-w does not have a GPIO pin enabled for the LED; instead it uses the wireless chip to extend the GPIOs - this would require some additional HAL code to interact with that chip.

phausler, 2024-06-13:
You can use the https://github.com/apple/swift-embedded-examples/tree/main/pico-w-blink-sdk example to do that instead. That one is built to use the existing pico sdk and interface for the pico-w

rauhul, 2024-06-28:
I think @phausler's solution is correct, please reopen this issue with more details if the solution does not work for you.

closed 2024-06-28 by rauhul