swiftlang/swift-embedded-examples #14: `nrfx-blink-sdk` can't build the example

naknut, 2024-06-12:
Hi! I have been following the instructions for the `nrfx-blink-sdk` sample but when I run `cmake --build build` I get the following error. From what I understand it seems like the `inttypes.h` header is not in the toolchain? It looks like its trying to use the development toolchain I downloaded from June 11.

```
[1/135] Preparing syscall dependency handling

[3/135] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.6.99 (/Users/naknut/zephyrproject/zephyr), build: v3.6.0-5909-gdfb240f37ced
[4/135] Generating _swiftcode.o
FAILED: _swiftcode.o /Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/build/_swiftcode.o 
cd /Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/build && /Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-06-11-a.xctoolchain/usr/bin/swiftc -target armv7em-none-none-eabi -Xcc -mfloat-abi=soft -Xcc -fshort-enums -Xfrontend -function-sections -enable-experimental-feature Embedded -wmo -parse-as-library $( echo '/Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/build/zephyr/include/generated/zephyr;/Users/naknut/zephyrproject/zephyr/include;/Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/build/zephyr/include/generated;/Users/naknut/zephyrproject/zephyr/soc/nordic;/Users/naknut/zephyrproject/zephyr/lib/libc/common/include;/Users/naknut/zephyrproject/zephyr/soc/nordic/nrf52/.;/Users/naknut/zephyrproject/zephyr/soc/nordic/common/.;/Users/naknut/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include;/Users/naknut/zephyrproject/zephyr/modules/cmsis/.;/Users/naknut/zephyrproject/modules/hal/nordic/nrfx;/Users/naknut/zephyrproject/modules/hal/nordic/nrfx/drivers/include;/Users/naknut/zephyrproject/modules/hal/nordic/nrfx/mdk;/Users/naknut/zephyrproject/zephyr/modules/hal_nordic/nrfx/.;/Users/naknut/zephyrproject/modules/debug/segger/SEGGER;/Users/naknut/zephyrproject/modules/debug/segger/Config;' | sed 's/ $//' | tr ';' '\n' | sed -e 's/\(.*\)/-Xcc -I\1/g' ) $( echo 'KERNEL;__ZEPHYR__=1;;__LINUX_ERRNO_EXTENSIONS__;PICOLIBC_LONG_LONG_PRINTF_SCANF;__PROGRAM_START;NRF52840_XXAA;K_HEAP_MEM_POOL_SIZE=0' | sed 's/ $/' | tr ';' '\n' | sed -e 's/\(.*\)/-Xcc \1/g' ) -Xcc -DKERNEL -Xcc -DNRF52840_XXAA -Xcc -DPICOLIBC_INTEGER_PRINTF_SCANF -Xcc -D_FORTIFY_SOURCE=1 -Xcc -D_POSIX_C_SOURCE=200809 -Xcc -D__LINUX_ERRNO_EXTENSIONS__ -Xcc -D__PROGRAM_START -Xcc -D__ZEPHYR__=1 -Xcc -I/opt/nordic/ncs/toolchains/20d68df7e5/opt/zephyr-sdk/arm-zephyr-eabi/picolibc/include -import-bridging-header /Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/BridgingHeader.h /Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/main.swift -c -o /Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/build/_swiftcode.o
sed: 1: "s/ $/": unterminated substitute in regular expression
error: generate-pch command failed with exit code 1 (use -v to see invocation)
/Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/BridgingHeader.h:14:10: note: in file included from /Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/BridgingHeader.h:14:
12 | #include <autoconf.h>
13 | 
14 | #include <zephyr/kernel.h>
   |          `- note: in file included from /Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/BridgingHeader.h:14:
15 | #include <zephyr/drivers/gpio.h>
16 | 

/Users/naknut/zephyrproject/zephyr/include/zephyr/kernel.h:17:10: note: in file included from /Users/naknut/zephyrproject/zephyr/include/zephyr/kernel.h:17:
  15 | 
  16 | #if !defined(_ASMLANGUAGE)
  17 | #include <zephyr/kernel_includes.h>
     |          `- note: in file included from /Users/naknut/zephyrproject/zephyr/include/zephyr/kernel.h:17:
  18 | #include <errno.h>
  19 | #include <limits.h>

/Users/naknut/zephyrproject/zephyr/include/zephyr/kernel_includes.h:34:10: note: in file included from /Users/naknut/zephyrproject/zephyr/include/zephyr/kernel_includes.h:34:
32 | #include <zephyr/kernel_structs.h>
33 | #include <zephyr/kernel_version.h>
34 | #include <zephyr/syscall.h>
   |          `- note: in file included from /Users/naknut/zephyrproject/zephyr/include/zephyr/kernel_includes.h:34:
35 | #include <zephyr/sys/printk.h>
36 | #include <zephyr/arch/cpu.h>

/Users/naknut/zephyrproject/zephyr/include/zephyr/syscall.h:11:10: error: 'zephyr/syscall_list.h' file not found
  9 | #define ZEPHYR_INCLUDE_SYSCALL_H_
 10 | 
 11 | #include <zephyr/syscall_list.h>
    |          `- error: 'zephyr/syscall_list.h' file not found
 12 | #include <zephyr/arch/syscall.h>
 13 | #include <stdbool.h>

/Users/naknut/zephyrproject/zephyr/include/zephyr/sys/printk.h:14:10: note: while building module '_Builtin_inttypes' imported from /Users/naknut/zephyrproject/zephyr/include/zephyr/sys/printk.h:14:
12 | #include <stddef.h>
13 | #include <stdarg.h>
14 | #include <inttypes.h>
   |          `- note: while building module '_Builtin_inttypes' imported from /Users/naknut/zephyrproject/zephyr/include/zephyr/sys/printk.h:14:
15 | 
16 | #ifdef __cplusplus

<module-includes>:1:10: note: in file included from <module-includes>:1:
1 | #include "inttypes.h"
  |          `- note: in file included from <module-includes>:1:
2 | 

2 errors generated.
<unknown>:0: error: failed to emit precompiled header '/var/folders/ls/h7hs2hk92g7_qynprm35gyqw0000gn/T/TemporaryDirectory.DbOdPp/BridgingHeader.pch' for bridging header '/Users/naknut/Developer/swift-embedded-examples/nrfx-blink-sdk/BridgingHeader.h'
/Library/Developer/Toolchains/swift-DEVELOPMENT-SNAPSHOT-2024-06-11-a.xctoolchain/usr/lib/swift/clang/include/inttypes.h:21:15: error: 'inttypes.h' file not found
19 | #endif
20 | 
21 | #include_next <inttypes.h>
   |               `- error: 'inttypes.h' file not found
22 | 
23 | #if defined(_MSC_VER) && _MSC_VER < 1900
[5/135] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
ninja: build stopped: subcommand failed.
```

rauhul, 2024-06-13:
reproduced this!

jmarcelino, 2024-06-14:
I'm getting an error even before that, from
https://github.com/apple/swift-embedded-examples/blob/2b93644fc4cca46c48e8da3023746c8d9e26007e/nrfx-blink-sdk/CMakeLists.txt#L15

`sed 's/;$$/'` seems it's missing the closing?  And `sed -e 's/\\\(.*\\\)/-Xcc \\1/g' \` is missing a -D ?

rauhul, 2024-06-15:
I'm hoping to replace the cmake in this example with the native CMake Swift support added in 3.29. @etcwilde and I started this today and I'll hopefully find time next week to finish it up

rauhul, 2024-06-21:
@naknut @jmarcelino could you try the `update-nrfx-cmake` branch of this repo? It should resolve the issue

naknut, 2024-06-21:
Yeah it's building now! Can't find my board to try and flash it though, but I can verify that it builds! Thanks!

jmarcelino, 2024-06-22:
Thanks so much @rauhul the example compiles well now but for some reason it's crashing my board, even JLInk stops after I upload the firmware. Fortunately running nrfjprog --recover in a loop while powering cycling get things back under control - but usually a sign the firmware is misbehaving.

I'm not using the NRF52840-DK though I'm using the rak4631 which is a nRF52840 + LoRa, but it is a board  supported by Zephyr and the Zephyr Blinky works well. led0 is correctly defined to be the user led.

Will try to figure out more over the next few days. 

Getting it to compile cleanly is already great, I'm quite new to this tooling. Thanks for that.

rauhul, 2024-06-22:
Maybe @kubamracek has some ideas but I unfortunately dont have an actual board to test against, so any deductions you're able to come to would be super helpful.

kubamracek, 2024-06-22:
I'll test this out on Monday

naknut, 2024-06-23:
I just found my board and tried to run it. It seems to be crashing for me too. Its gets written to the board successfully but the LED never starts blinking. This is done on a nRF52840-DK

bobbelkl, 2024-07-01:
I have the same issue. It builds, but doesn't execute anything on the board. 
Not sure if it's just my machine but $TOOLCHAINS is being ignored and I had to link the toolchain in the CMakeList file:
`set(CMAKE_Swift_COMPILER /Library/Developer/Toolchains/swift-latest.xctoolchain/usr/bin/swiftc)`
I see two warnings that caught my eye:
```
[133/140] Linking Swift static library libapp_swift.a
warning: /Applications/Xcode15.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: archive library: libapp_swift.a the table of contents is empty (no object file members in the library define global symbols)
[134/140] Linking Swift static library app/libapp.a
warning: /Applications/Xcode15.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: archive library: app/libapp.a the table of contents is empty (no object file members in the library define global symbols)
```
No clue why it defaults to Xcode, but there is no ranlib in the last development toolchain.

kubamracek, 2024-07-09:
I was able to produce the problem where the example wouldn't actually start blinking the LED. Fixing that as part of https://github.com/apple/swift-embedded-examples/pull/32.

closed 2024-07-09 by kubamracek