swiftlang/swift-embedded-examples #16: Windows Support

dlbuckley, 2024-06-14:
I tried to play around with the Pico SDK blink example last night from my windows machine using the latest Swift 6 toolchain and didn't have much luck. I'm not particularly proficient in cmake but it seems like some modifications are required to allow for Windows support using 'swift build' since it seems to be using 'xcrun'. (I would use macOS but I only have a work machine and it's USB ports have been locked down)

I had a look through the various docs regarding embedded Swift and it all seems to be macOS based so far. Are there any docs out there, or plans for docs, for what modifications would be needed for Windows/Linux?

joseph-montanez, 2024-06-14:
I've gotten this far at least on Linux: 

`<unknown>:0: error: unable to load standard library for target 'armv6m-none-none-eabi'`

This does not work, unless I add `-enable-experimental-feature Embedded -wmo` the issue is that then the main annotation no longer works and I get:
```
/home/joseph/Documents/swift-embedded-examples/pico-w-blink-sdk/Main.swift:12:1: error: 'main' attribute cannot be used in a module that contains top-level code
10 | //===----------------------------------------------------------------------===//
11 | 
12 | @main
   | |- error: 'main' attribute cannot be used in a module that contains top-level code
   | |- note: top-level code defined in this source file
   | `- note: pass '-parse-as-library' to compiler invocation if this is intentional
13 | struct Main {
14 |     static func main() {
```

If I remove main, then it will compile as a **library**. I tried as both main development branch and 6.0 development branch, both produce the same issue.  Use -enable-experimental-feature Embedded means I cannot compile as an isolate executable, and instead its a library and I have to pull into C.

BridgingHeader.h
```c
//===----------------------------------------------------------------------===//
//
// This source file is part of the Swift open source project
//
// Copyright (c) 2024 Apple Inc. and the Swift project authors.
// Licensed under Apache License v2.0 with Runtime Library Exception
//
// See https://swift.org/LICENSE.txt for license information
//
//===----------------------------------------------------------------------===//

#pragma once

// Define __unused attribute if not already defined
#define __unused __attribute__((unused))

// Define PICO_DEFAULT_UART if not already defined
#ifndef PICO_DEFAULT_UART
#define PICO_DEFAULT_UART 0
#endif

#ifndef PICO_DEFAULT_UART_INSTANCE
#define PICO_DEFAULT_UART_INSTANCE uart0
#endif

// Include Pico SDK headers with proper guards
#ifndef PICO_STDLIB_H
#define PICO_STDLIB_H
#include "pico/stdlib.h"
#endif

#ifndef PICO_CYW43_ARCH_H
#define PICO_CYW43_ARCH_H
#include "pico/cyw43_arch.h"
#endif

// Check if the macros are already defined
#ifndef LITTLE_ENDIAN
#define LITTLE_ENDIAN 1234
#endif

#ifndef BIG_ENDIAN
#define BIG_ENDIAN 4321
#endif

#ifndef BYTE_ORDER
#define BYTE_ORDER LITTLE_ENDIAN
#endif
```
[... 458 lines cut ...]

/home/joseph/Documents/swift-embedded-examples/pico-sdk/lib/lwip/src/include/lwip/arch.h:67:9: note: previous definition is here
 65 |  */
 66 | #ifndef BYTE_ORDER
 67 | #define BYTE_ORDER LITTLE_ENDIAN
    |         `- note: previous definition is here
 68 | #endif
 69 | 
/home/joseph/Apps/swift-DEVELOPMENT-SNAPSHOT-2024-06-13-a-ubuntu22.04/usr/bin/swift-frontend -frontend -c -primary-file /home/joseph/Documents/swift-embedded-examples/pico-w-blink-sdk/Main.swift -target armv6m-none-none-eabi -disable-objc-interop -function-sections -empty-abi-descriptor -Xcc -working-directory -Xcc /home/joseph/Documents/swift-embedded-examples/pico-w-blink-sdk/build -resource-dir /home/joseph/Apps/swift-DEVELOPMENT-SNAPSHOT-2024-06-13-a-ubuntu22.04/usr/lib/swift -Xcc -mfloat-abi=soft -Xcc -fshort-enums -Xcc -DCYW43_LWIP -Xcc -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND -Xcc -DPICO_DEFAULT_UART=0 -Xcc -DPICO_BOARD=pico_w -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/lib/lwip/src/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/lib/cyw43-driver/src -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/pico_async_context/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/pico_rand/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/pico_lwip/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/pico_cyw43_driver/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/pico_stdio/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/hardware_base/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/hardware_timer/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/hardware_gpio/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/hardware_irq/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/hardware_uart/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/common/pico_base/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/common/pico_stdlib/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/common/pico_time/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/pico_platform/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/boot_stage2/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2040/hardware_regs/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2040/hardware_structs/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-sdk/src/rp2_common/pico_cyw43_arch/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-w-blink-sdk/include -Xcc -I/home/joseph/Documents/swift-embedded-examples/pico-w-blink-sdk/build/generated/pico_base -Xcc -I/usr/include -Xcc -I/usr/include/arm-none-eabi -import-objc-header /tmp/TemporaryDirectory.EwsOWU/BridgingHeader.pch -modu [... cut ...]
[3/10] Building CXX object CMakeFiles/pioasm.dir/main.cpp.o
[4/10] Building CXX object CMakeFiles/pioasm.dir/python_output.cpp.o
[5/10] Building CXX object CMakeFiles/pioasm.dir/hex_output.cpp.o
[6/10] Building CXX object CMakeFiles/pioasm.dir/gen/lexer.cpp.o
[7/10] Building CXX object CMakeFiles/pioasm.dir/c_sdk_output.cpp.o
[8/10] Building CXX object CMakeFiles/pioasm.dir/gen/parser.cpp.o
[9/10] Building CXX object CMakeFiles/pioasm.dir/pio_assembler.cpp.o
[10/10] Linking CXX executable pioasm
ninja: build stopped: subcommand failed.

joseph-montanez, 2024-06-15:
I did a bit more digging, I think the issue is around `TOOLCHAINS` environmental, this is a macOS only environmental that's taken into account when building. Meaning on Linux and Windows it will be ignored.

https://github.com/apple/swift/blob/219b4a61d4c80e6e67338cad51b22db27f79723e/utils/build-script#L675

The toolchains points to STDLIB and there is where the errors happens:
https://github.com/apple/swift/blob/219b4a61d4c80e6e67338cad51b22db27f79723e/lib/Frontend/Frontend.cpp#L1530

I can get this to build on macOS with the code in the repo as long as I add:
```cmake
set(CMAKE_ASM_COMPILER "arm-none-eabi-as")
set(CMAKE_ASM_COMPILER_ID GNU)
set(CMAKE_ASM_FLAGS "-march=armv6-m")
```

On Linux I tried setting `--tools-directory` for **swiftc**, but nothing I set it to fixed the error

joseph-montanez, 2024-06-15:
Here is the final version for Linux (Ubuntu 24.04) that I got working:

```bash
export PICO_BOARD=pico_w
export PICO_SDK_PATH=/home/joseph/Documents/swift-embedded-examples/pico-sdk
export PICO_TOOLCHAIN_PATH=/home/joseph/Apps/arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi/bin

cmake -B build -G Ninja .
cmake --build build
```

Bash commands for Swift's docker image (Ubuntu 22.04):
```bash
docker pull swiftlang/swift:nightly-6.0-jammy
docker run --privileged --interactive --tty --name swift-latest -v $(pwd):/mnt swiftlang/swift:nightly-6.0-jammy /bin/bash

cd /mnt/
apt-get update && apt-get install wget gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib gcc-multilib ninja-build python3 xz-utils
wget https://github.com/Kitware/CMake/releases/download/v3.29.0/cmake-3.29.0-linux-x86_64.sh
./cmake-3.29.0-linux-x86_64.sh --skip-license --prefix=/usr
git clone https://github.com/raspberrypi/pico-sdk.git
wget "https://developer.arm.com/-/media/Files/downloads/gnu/13.2.rel1/binrel/arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi.tar.xz?rev=e434b9ea4afc4ed7998329566b764309&hash=CA590209F5774EE1C96E6450E14A3E26" -o arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi.tar.xz
xz -d arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi.tar.xz
tar -xvf arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eabi.tar

export PICO_BOARD=pico_w
export PICO_SDK_PATH=/mnt/pico-sdk
export PICO_TOOLCHAIN_PATH=/mnt/arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi/bin
export TOOLCHAINS=/mnt/swift-DEVELOPMENT-SNAPSHOT-2024-06-13-a-ubuntu22.04/usr

cmake -B build -G Ninja .
cmake --build build
```

These are the files I've changed

**CMakeLists.txt**
```cmake
cmake_minimum_required(VERSION 3.13)

set(CMAKE_ASM_COMPILER "arm-none-eabi-as")
set(CMAKE_ASM_COMPILER_ID GNU)

# Set the Swift compiler for CMake
find_program(SWIFTC swiftc REQUIRED)
set(CMAKE_Swift_COMPILER ${SWIFTC})

set(CMAKE_Swift_COMPILER_FORCED TRUE)

set(PICO_ENABLE_TESTS off)
set(PICO_BOARD "pico_w")

# Manually specified include directories
set(MANUAL_INCLUDE_DIRECTORIES
        $ENV{PICO_SDK_PATH}/lib/lwip/src/include
        $ENV{PICO_SDK_PATH}/lib/cyw43-driver/src
        $ENV{PICO_SDK_PATH}/src/rp2_common/pico_async_context/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/pico_rand/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/pico_lwip/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/pico_cyw43_driver/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/pico_stdio/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/hardware_base/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/hardware_timer/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/hardware_gpio/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/hardware_irq/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/hardware_uart/include
        $ENV{PICO_SDK_PATH}/src/common/pico_base/include
        $ENV{PICO_SDK_PATH}/src/common/pico_stdlib/include
        $ENV{PICO_SDK_PATH}/src/common/pico_time/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/pico_platform/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/boot_stage2/include
        $ENV{PICO_SDK_PATH}/src/rp2040/hardware_regs/include
        $ENV{PICO_SDK_PATH}/src/rp2040/hardware_structs/include
        $ENV{PICO_SDK_PATH}/src/rp2_common/pico_cyw43_arch/include
        ${CMAKE_CURRENT_LIST_DIR}/include
        ${CMAKE_CURRENT_BINARY_DIR}/generated/pico_base
)

list(APPEND PICO_INCLUDE_DIRECTORIES ${MANUAL_INCLUDE_DIRECTORIES})

message(STATUS "PICO Include directories: ${PICO_INCLUDE_DIRECTORIES}")

set(INCLUDE_DIRS "")
foreach(dir ${PICO_INCLUDE_DIRECTORIES})
    list(APPEND INCLUDE_DIRS -Xcc -I${dir})
endforeach()
list(APPEND INCLUDE_DIRS -Xcc -I/usr/include -Xcc -I/usr/include/arm-none-eabi)

message(STATUS "Include directories: ${INCLUDE_DIRS}")

if(CMAKE_CXX_COMPILER_ID MATCHES "Clang")
    add_compile_definitions(__unused=__attribute__((unused)))
endif()

include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)

project(swift-blinky LANGUAGES C CXX ASM Swift)

pico_sdk_init()

add_executable(swift-blinky
        BridgingHeader.h)

#-no-allocations -Osize -Xlinker -dead_strip
add_custom_command(
        OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
        COMMAND
        ${SWIFTC}
        -target armv6m-none-none-eabi
        -Xcc -mfloat-abi=soft
        -Xcc -fshort-enums
        -Xcc -DCYW43_LWIP
        -Xcc -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND
        -Xcc -DPICO_DEFAULT_UART=0
        -Xcc -DPICO_BOARD=pico_w
        -Xfrontend -function-sections -enable-experimental-feature Embedded -wmo -parse-as-library
        ${INCLUDE_DIRS}
        -import-bridging-header ${CMAKE_CURRENT_LIST_DIR}/BridgingHeader.h
        ${CMAKE_CURRENT_LIST_DIR}/Main.swift
        -v
        -c -o ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
        DEPENDS
        ${CMAKE_CURRENT_LIST_DIR}/BridgingHeader.h
        ${CMAKE_CURRENT_LIST_DIR}/Main.swift
)

add_custom_target(swift-blinky-swiftcode DEPENDS ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o)

target_link_libraries(swift-blinky
        pico_stdlib hardware_uart hardware_gpio pico_lwip_arch pico_cyw43_arch_none
        ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
)

add_dependencies(swift-blinky swift-blinky-swiftcode)
pico_add_extra_outputs(swift-blinky)
```

**BridgingHeader.h**
```c
#pragma once

// Define __unused attribute if not already defined
#define __unused __attribute__((unused))

// Define PICO_DEFAULT_UART if not already defined
#ifndef PICO_DEFAULT_UART
#define PICO_DEFAULT_UART 0
#endif

#ifndef PICO_DEFAULT_UART_INSTANCE
#define PICO_DEFAULT_UART_INSTANCE uart0
#endif

#include "pico/stdlib.h"
#include "pico/cyw43_arch.h"
```

shivarajd, 2024-06-27:
I have created a simple Docker-based setup for developing embedded Swift applications for the Raspberry Pi Pico and Pico W microcontrollers using Pico-SDK. I have tested it on macOS, Ubuntu, and Windows host.  https://github.com/shivarajd/swift-embedded-pico-docker