swiftlang/swift-embedded-examples #17: esp32-led-strip-sdk unable to build in VS Code using PlatformIO

oleksandr-kosylov, 2024-06-16:
I'm configuring CMakeLists same as described in Example, it successfully works from Terminal using idf.py build BUT doesn't work from in VSCode using PlatformIO (idf 5.2.1)

Looks like custom command/custom target never gets called, so _swiftcode.o never generated and build fails due to undefined symbols for app_main function which is implemented in Main.swift.

I see command and target configuration in build.ninja but they are not called, I've tied to replace swiftc command with custom test command but it's not getting executed.

I've either filed an issue in PlatformIO repo either: https://github.com/platformio/platform-espressif32/issues/1401

Please advice.

avkghost, 2024-06-17:
I have a similar problem.
As far as I understand in my case, the problem is the incorrect path for a Swift compiler.
When I changed the path (exportSWIFTC, or replaced the actual path) with a real path to the toolchain (Swift 6, nightly build) I almost compiled the project without the final binary because of errors, see below.

`
[home]/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: The gap between .flash.appdesc and .flash.rodata must not exist to produce the final bin image.
[home]/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: The gap between .flash.rodata and .eh_frame_hdr must not exist to produce the final bin image.
collect2: error: ld returned 1 exit status
`

@kubamracek Could you please help us to solve the issue?
Thanks for the beautiful presentation.

kubamracek, 2024-06-17:
I think these might be separate problems. If you can, post full logs and also details about your OS, configuration, environment, etc.

If swiftc is not getting called at all, my best guess would be that CMake is too old. The Swift support in CMake is only available in CMake 3.29+ and it's conceivable that in a terminal your PATH is pointing to a different CMake than from VSCode. Full build logs, maybe even screenshots, would be helpful. Also, what is your declared minimum CMake version in your CMakeListst.txt?

> As far as I understand in my case, the problem is the incorrect path for a Swift compiler.

Right, this is probably a common issue. You can force a Swift compiler path with `-DCMAKE_Swift_COMPILER=<path>`.

```
[home]/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: The gap between .flash.appdesc and .flash.rodata must not exist to produce the final bin image. [home]/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: The gap between .flash.rodata and .eh_frame_hdr must not exist to produce the final bin image. collect2: error: ld returned 1 exit status
```

What's your ESP IDF version? And are you getting this error just by building the `esp32-led-strip-sdk` example code unmodified?

avkghost, 2024-06-17:
@kubamracek 

The system is Ubuntu 24.04.
> If swiftc is not getting called at all, my best guess would be that CMake is too old. The Swift support in CMake is only available in CMake 3.29+
Looks like the problem belongs to my version of Cmake (3.28.3-1build7) 

> What's your ESP IDF version? 
ESP-IDF v5.4-dev-826-g8760e6d2a7-dirty. However, I'm going to compile the example with the latest stable SDK (5.2.2).  

> And are you getting this error just by building the esp32-led-strip-sdk example code unmodified?
Yes.

Akeyzerr, 2024-06-17:
I'm reproducing 
`.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: The gap between .flash.rodata and .eh_frame_hdr must not exist to produce the final bin image.`
`ninja failed with exit code 1, output of the command is in the .../build/log/idf_py_stderr_output_62229 and .../build/log/idf_py_stdout_output_62229`
on a MacMini M1 w/latest Sonoma, IDF v5.4-dev-826-g8760e6d2a7 and Apple Swift version 6.0-dev (LLVM 57177aa1b91540b, Swift 8be62863326595c)
Target: arm64-apple-macosx14.0

erhankur, 2024-06-17:
@avkghost You can apply the attached patch to fix newlib getentropy error. 
[0001-newlib-getentropy-implementation.patch](https://github.com/user-attachments/files/15877608/0001-newlib-getentropy-implementation.patch)

avkghost, 2024-06-17:
@erhankur Thanks. It's help.
 
 Maybe you could help with the problem with ESP-IDF v5.4-dev-826-g8760e6d2a7-dirty
 
`.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: The gap between .flash.rodata and .eh_frame_hdr must not exist to produce the final bin image. collect2: error: ld returned 1 exit status`

erhankur, 2024-06-17:
As a temporary solution you can comment out 2 assertions from `components/esp_system/ld/esp32c6/sections.ld.in`

```
-  ASSERT_SECTIONS_GAP(.flash.appdesc, .flash.rodata)
+  /* ASSERT_SECTIONS_GAP(.flash.appdesc, .flash.rodata) */

- ASSERT_SECTIONS_GAP(.flash.rodata, .eh_frame_hdr)
+  /* ASSERT_SECTIONS_GAP(.flash.rodata, .eh_frame_hdr) */

```

avkghost, 2024-06-17:
@erhankur Thank you so much.

erhankur, 2024-06-17:
The current locations of `.swift_modhash` , `.got ` and `.got.plt ` sections cause the assert errors. 

```
  [13] .flash.text       PROGBITS        42000020 010020 016768 00  AX  0   0  2
  [14] .flash_rodat[...] NOBITS          42000020 027020 018000 00  WA  0   0  1
  [15] .flash.appdesc    PROGBITS        42018020 03f020 000100 00   A  0   0 16
  [16] .swift_modhash    PROGBITS        42018120 03f120 000010 00  Ao  0   0  1
  [17] .flash.rodata     PROGBITS        42018130 03f130 00abdc 00  WA  0   0 16
  [18] .got              PROGBITS        42022d0c 049d0c 000008 04  WA  0   0  4
  [19] .got.plt          PROGBITS        42022d14 049d14 000008 04  WA  0   0  4
  [20] .eh_frame_hdr     PROGBITS        42022d1c 049d1c 000000 00   W  0   0  1
  [21] .eh_frame         PROGBITS        42022d1c 049d1c 000000 00   W  0   0  1
  [22] .flash.tdata      PROGBITS        42022d1c 049d1c 000000 00   W  0   0  1
  ```
  
This patch also fixes that errors.

[0001-move-swift-sections-to-proper-memory-regions.patch](https://github.com/user-attachments/files/15878265/0001-move-swift-sections-to-proper-memory-regions.patch)

georgik, 2024-06-18:
The solution works for C6.
The similar correction would be required also for C3.
It's possible to test the build process simply by changing target:
```
idf.py set-target esp32c3
idf.py build
```

PaulSolt, 2024-09-04:
@erhankur 

1. Where am I applying the patch?

2. I couldn't seem to find the assertions in `components/esp_system/ld/esp32c6/sections.ld.in`

3. I'm new to embedded programming, what does this mean?

> The current locations of .swift_modhash , .got  and .got.plt  sections cause the assert errors.

PaulSolt, 2024-09-12:
@erhankur Is this an issue that will be fixed in https://github.com/espressif/esp-idf ?

erhankur, 2024-09-12:
@PaulSolt try using the idf master branch. You shouldn't need to apply a patch.

PaulSolt, 2024-09-12:
@erhankur What release will have the fixes from `master`?

The previous patches didn't work for me, so I had to manually apply the changes. Attached are the patches based on your patches that worked for me.

1. Download the patch you need to fix random numbers
	1. [esp32c6_random_crash_fix_5.2.2.patch](https://github.com/user-attachments/files/16984049/esp32c6_random_crash_fix_5.2.2.patch)
	2. [esp32c6_random_crash_fix_5.3.1.patch](https://github.com/user-attachments/files/16984054/esp32c6_random_crash_fix_5.3.1.patch)
2. Navigate to your ESP build folder
	```bash
	cd ~/esp/esp-idf/
	```
3. Apply the patch (5.2.2 or 5.3.1)
	```bash
	git apply ~/Dowloads/esp32c6_random_crash_fix_5.2.2.patch
	```
4. Navigate back to your project and build. It should pick up the changes and fix your problem.
    ```bash
    idf.py build
    ```

erhankur, 2024-09-13:
@PaulSolt Please stay in the master (v5.4) branch. Don't checkout to the release v5.2 or v5.3

PaulSolt, 2024-09-30:
Thank you @erhankur for the details. I'll switch to the master (v5.4) branch and recommend others do the same. I appreciate your help.