swiftlang/swift-embedded-examples #19: esp32-p4 support - march=rv32imafc

georgik, 2024-06-21:
The build of ESP32 Led strip example for ESP32-P4 which is RISC-V IMACF requires:

1. Correction in linker mentioned in https://github.com/apple/swift-embedded-examples/issues/17#issuecomment-2174534399

2. Correction of build script, because of linking problem:

```
esp-idf/main/_swiftcode.o: can't link soft-float modules with single-float modules
failed to merge target specific data of file esp-idf/main/_swiftcode.o
```

The solution is to add `-Xcc -march=rv32imafc -Xcc -mabi=ilp32f` to the swiftc command line to ensure the Swift compiler uses the same architecture and ABI as the C and C++ compilers.
Plus, add `-Xlinker -march=rv32imafc -Xlinker -mabi=ilp32f` to pass these flags to the linker.

Updated `main/CMakeLists.txt` look like this:
```cmake
idf_component_register(
    SRCS "Dummy.c"
    INCLUDE_DIRS "."
)

execute_process(COMMAND xcrun -f swiftc OUTPUT_VARIABLE SWIFTC OUTPUT_STRIP_TRAILING_WHITESPACE)

add_custom_command(
    OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
    COMMAND
        ${SWIFTC}
        -target riscv32-none-none-eabi
        -Xfrontend -function-sections -enable-experimental-feature Embedded -wmo -parse-as-library -Osize
        -Xcc -march=rv32imafc -Xcc -mabi=ilp32f
        -Xlinker -march=rv32imafc -Xlinker -mabi=ilp32f
        $$\( echo '$<TARGET_PROPERTY:__idf_main,INCLUDE_DIRECTORIES>' | tr '\;' '\\n' | sed -e 's/\\\(.*\\\)/-Xcc -I\\1/g' \)
        $$\( echo '${CMAKE_C_IMPLICIT_INCLUDE_DIRECTORIES}'           | tr ' '  '\\n' | sed -e 's/\\\(.*\\\)/-Xcc -I\\1/g' \)
        -import-bridging-header ${CMAKE_CURRENT_LIST_DIR}/BridgingHeader.h
        ${CMAKE_CURRENT_LIST_DIR}/Main.swift
        ${CMAKE_CURRENT_LIST_DIR}/LedStrip.swift
        -c -o ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
    DEPENDS
        ${CMAKE_CURRENT_LIST_DIR}/BridgingHeader.h
        ${CMAKE_CURRENT_LIST_DIR}/Main.swift
        ${CMAKE_CURRENT_LIST_DIR}/LedStrip.swift
)
add_custom_target(main-swiftcode DEPENDS ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o)

target_link_libraries(__idf_main
    ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
)
add_dependencies(__idf_main main-swiftcode)
```

rauhul, 2024-06-21:
Is there a way we can adjust this Cmake file to support both boards?

erhankur, 2024-06-24:
Quick solution;

```cmake
idf_component_register(
    SRCS "Dummy.c"
    INCLUDE_DIRS "."
)

idf_build_get_property(target IDF_TARGET)
idf_build_get_property(arch IDF_TARGET_ARCH)

if("${arch}" STREQUAL "xtensa")
    message(FATAL_ERROR "Target not supported ${target}")
endif()

if(${target} STREQUAL "esp32c2" OR ${target} STREQUAL "esp32c3")
    set(march_flag "rv32imc_zicsr_zifencei")
    set(mabi_flag "ilp32")
elseif(${target} STREQUAL "esp32p4")
    set(march_flag "rv32imafc_zicsr_zifencei")
    set(mabi_flag "ilp32f")
else()
    set(march_flag "rv32imac_zicsr_zifencei")
    set(mabi_flag "ilp32")
endif()

#execute_process(COMMAND xcrun -f swiftc OUTPUT_VARIABLE SWIFTC OUTPUT_STRIP_TRAILING_WHITESPACE)
set(SWIFTC $ENV{TOOLCHAINS}/usr/bin/swiftc)

add_custom_command(
    OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
    COMMAND
        ${SWIFTC}
        -target riscv32-none-none-eabi
        -Xfrontend -function-sections -enable-experimental-feature Embedded -static -wmo -parse-as-library -Osize
        -Xcc -march=${march_flag} -Xcc -mabi=${mabi_flag}
        -Xlinker -march=${march_flag} -Xlinker -mabi=${mabi_flag}
        .....
```

rauhul, 2024-06-28:
@erhankur would you be interested in submitting a PR which includes these changes?

PR #25 (2024-07-03): esp: add march and mabi flags for all riscv chips
A better solution could be to get the March and Mabi flags from the idf `toolchain-clang-<target>.cmake` file. However, for now, the IDF build uses the GCC toolchain instead of Clang. I would like to test the Clang build and then modify the related parts.

Closes #19
closed 2024-07-03 by rauhul