swiftlang/swift-embedded-examples #22: `esp32-led-strip-sdk` crash on ESP32-C6 , cannot use `.random`

gerwindehaan, 2024-06-28:
I experience a crash/bootloop on the `esp32-led-strip-sdk` example when run on my ESP32-C6

I traced it down to the use of `.random` in the led strip color generation (it works if I avoid that). The behavior is illustrated by the following minimal snippet:

```swift
@_cdecl("app_main")
func app_main() {
  print("Running Int.random")  
  let _ = Int.random(in: 0...16)
  print(". ")
  while(true){ }
}
```
 
Running this on device causes a crash when the `.random` is invoked, followed by a reboot. When compiling this sample (or anything where `.random` is spawned, a _warning is spawned around _ `_getentropy`:

```
[5/7] Linking CXX executable main.elf
/Users/gerwindehaan/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240305/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/gerwindehaan/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240305/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/lib/rv32imac_zicsr_zifencei/ilp32/no-rtti/libc.a(libc_a-getentropyr.o): in function `_getentropy_r':
/builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin21.1/riscv32-esp-elf/src/newlib/newlib/libc/reent/getentropyr.c:48:(.text+0x16): warning: _getentropy is not implemented and will always fail
```

I am wondering if this has to do with my configuration of the device, or that I should somehow enable the [random generator RNG](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/random.html) when Wifi/BLE are off .

erhankur, 2024-06-28:
We have shared the patch here https://github.com/apple/swift-embedded-examples/issues/17#issuecomment-2174479217

gerwindehaan, 2024-06-28:
Thanks, after applying this patch to my `esp-idf` source solves the issue with `.random` and the sample works as advertised. Closing as this is tracked by https://github.com/apple/swift-embedded-examples/issues/17 and https://github.com/apple/swift-matter-examples/issues/5

closed 2024-06-28 by gerwindehaan