swiftlang/swift-embedded-examples #38: Using strings causes linker failure in nrfx-blink-sdk

nrbrook, 2024-07-29:
Adding `let a = "abc"` to Main.swift causes linker failure, without any clear error message:

> % cmake --build build                                                 
> [3/7] Linking C executable zephyr/zephyr_pre0.elf
> FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /Users/nick/Documents/Repos/swift-embedded-examples/nrfx-blink-sdk/build/zephyr/zephyr_pre0.map 
> : && /opt/nordic/ncs/toolchains/580e4ef81c/opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -g -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/Users/nick/Documents/Repos/swift-embedded-examples/nrfx-blink-sdk/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/soc/soc/arm/nordic_nrf/libsoc__arm__nordic_nrf.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/nrf/lib/fatal_error/lib..__nrf__lib__fatal_error.a  modules/nrf/drivers/hw_cc3xx/lib..__nrf__drivers__hw_cc3xx.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/segger/libmodules__segger.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/nordic/ncs/toolchains/580e4ef81c/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -L/Users/nick/Documents/Repos/swift-embedded-examples/nrfx-blink-sdk/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  /opt/nordic/ncs/v2.6.1/nrfxlib/crypto/nrf_cc310_platform/lib/cortex-m4/soft-float/no-interrupts/libnrf_cc310_platform_0.9.19.a  --specs=picolibc.specs  -lc  -lgcc && cd /Users/nick/Documents/Repos/swift-embedded-examples/nrfx-blink-sdk/build/zephyr && /opt/homebrew/Cellar/cmake/3.30.1/bin/cmake -E true
> /opt/nordic/ncs/toolchains/580e4ef81c/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: warning: orphan section `.swift_modhash' from `app/libapp.a(Main.swift.obj)' being placed in section `.swift_modhash'
> /opt/nordic/ncs/toolchains/580e4ef81c/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: warning: orphan section `.debug_names' from `app/libapp.a(Main.swift.obj)' being placed in section `.debug_names'
> collect2: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.

I'm aware this might require allocation, which the project does not currently support. I have had a go at implementing `posix_memalign` but can't get far enough to test it...

```C
#include <stddef.h>
#include <zephyr/kernel.h>

int posix_memalign(void **memptr, size_t alignment, size_t size) {
    void *ptr = k_aligned_alloc(alignment, size);
    if (ptr == NULL) {
        return ENOMEM;
    }
    *memptr = ptr;
    return 0;
}
```

kubamracek, 2024-08-02:
I can reproduce this problem. Somehow the linker is failing to link but not saying why (the warning about the orphan section is present even in the successful case so I don’t think that’s a real problem). Let me dig in.

nrbrook, 2024-08-02:
Thanks. I enjoyed the WWDC video by the way. Excited to start building for embedded devices using Swift!

closed 2024-08-16 by kubamracek