swiftlang/swift-embedded-examples #42: Support for Arduino libraries?

maartin0, 2024-08-02:
I'm trying to get [arduino-esp32](https://github.com/espressif/arduino-esp32/) to compile using the [esp32-led-blink-sdk](https://github.com/apple/swift-embedded-examples/tree/main/esp32-led-blink-sdk) example as a template, which would be amazing if it worked.

I'm pretty sure it shouldn't be super difficult given how it's literally just a C++ library, but the only issue I've run into is that `swiftc` doesn't seem to like the preprocessor macros that Arduino uses to handle different architectures etc.

Here's the bit of [arduino's CMakeLists.txt](https://github.com/espressif/arduino-esp32/blob/70786dc5fa51601f525496d0f92a220c917b4ad9/CMakeLists.txt#L317-L323) that defines the compiler flags:
```cmake
target_compile_options(${COMPONENT_TARGET} PUBLIC
    -DARDUINO=10812
    -DARDUINO_${idf_target_for_macro}_DEV
    -DARDUINO_ARCH_ESP32
    -DARDUINO_BOARD="${idf_target_caps}_DEV"
    -DARDUINO_VARIANT="${CONFIG_ARDUINO_VARIANT}"
    -DESP32)
```

This is what the `swiftc` call then looks like:
```bash
/usr/bin/swiftc -j 8 -num-threads 8 -c 
-DESP_PLATFORM -DIDF_VER=\"v5.1.4\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS 
-parse-as-library -static -emit-module 
-emit-module-path esp-idf/main/__idf_main.swiftmodule
-module-name __idf_main 
-module-link-name main 
-wmo -color-diagnostics
-target riscv32-none-none-eabi
-Xfrontend -function-sections -enable-experimental-feature Embedded -wmo -parse-as-library -Osize 
-Xcc -march=rv32imc_zicsr_zifencei -Xcc -mabi=ilp32 
-pch-output-dir /tmp
-Xfrontend 
-enable-single-module-llvm-emission 
-Xcc -I... -Xcc -I....
-import-bridging-header main/BridgingHeader.h
-cxx-interoperability-mode=default
-DARDUINO=10812 -DARDUINO_ESP32C3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32C3_DEV\" -DARDUINO_VARIANT=\"esp32c3\" -DESP32 
-output-file-map esp-idf/main/CMakeFiles/__idf_main.dir//output-file-map.json 
-I ... -I ....
```

Which raises the following warnings:
```
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'IDF_VER="v5.1.4"')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'MBEDTLS_CONFIG_FILE="mbedtls/esp_config.h"')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'SOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'ARDUINO=10812')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'ARDUINO_BOARD="ESP32C3_DEV"')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'ARDUINO_VARIANT="esp32c3"')
```

Which is then used by a lot of Arduino libraries like [OneWire](https://github.com/PaulStoffregen/OneWire/blob/72249e22ef9092b0750e303d266199965a89e500/OneWire.h#L12-L17) for example ([list](https://github.com/search?q=%2F%5E%23if%20ARDUINO%20%3E%3D%2F%20path%3A*.h&type=code)):
```cpp
#if ARDUINO >= 100
#include <Arduino.h>       // for delayMicroseconds, digitalPinToBitMask, etc
#else
#include "WProgram.h"      // for delayMicroseconds
#include "pins_arduino.h"  // for digitalPinToBitMask, etc
#endif
```

Creating an error like this:
```cmake
<unknown>:0: error: failed to emit precompiled header '/tmp/BridgingHeader-swift_1PSIJ0LN6QHW8-clang_1Y9RGGHIW2TB2.pch' for bridging header 'main/BridgingHeader.h'
2 warnings and 4 errors generated.
components/OneWire/OneWire.h:15:10: error: 'WProgram.h' file not found
 13 | #include <Arduino.h>       // for delayMicroseconds, digitalPinToBitMask, etc
 14 | #else
 15 | #include "WProgram.h"      // for delayMicroseconds
    |          `- error: 'WProgram.h' file not found
 16 | #include "pins_arduino.h"  // for digitalPinToBitMask, etc
 17 | #endif
 ```

So it seems that the issue is `swiftc` only supports boolean preproccessor macros; I've tried putting the attributes in my `BridgingHeader.h` as `#define`s but that doesn't seem to work, and I don't know enough about swift's internals how to know what else to try?

Here's what is basically a MNWE (minimal not working example): https://github.com/maartin0/esp32-arduino-swift-test but I'm happy to test any ideas you might have so you don't have to set up a workspace (Arduino is like 1.5 GB)

rauhul, 2024-08-06:
You likely want to pass the defines with `-Xcc -DARDUINO=10812` so they go to the clang importer instead of the swift compiler

rauhul, 2024-08-06:
I haven never seen that symbol! @kubamracek have you?

maartin0, 2024-08-12:
I've figured out that exact error, it's because before [you updated](https://github.com/apple/swift-embedded-examples/commit/b9bc89abcb490cb801f8662d6df8f828aa6f43ad) the example to use CMake's native Swift support I had included `-cxx-interoperability-mode=default` in swiftc's arguments, and didn't remove it when I updated the rest of the file to try and use the same configuration as you. I guess I hadn't properly read the [docs](https://www.swift.org/documentation/cxx-interop/project-build-setup/#mixing-swift-and-c-using-other-build-systems) because it explicitly says that's for when you have a clang module map which I don't have since this is manually just `-I` including everything

That original [MNWE](https://github.com/maartin0/esp32-arduino-swift-test) now builds, but the original project that I was working on still doesn't compile because it doesn't seem like it's treating the `arduino` library as C++? e.g.
```
components/arduino/libraries/Wire/src/Wire.h:55:17: error: unknown type name 'public'
 53 | #endif /* SOC_I2C_SUPPORT_SLAVE */
 54 |
 55 | class TwoWire : public HardwareI2C {
    |                 `- error: unknown type name 'public'
 56 | protected:
 57 |   uint8_t num;
 ```
I've updated the MNWE's bridging header to `#include "Wire.h"` which gives it that same error now,

Reading those [docs](https://www.swift.org/documentation/cxx-interop/project-build-setup/#mixing-swift-and-c-using-other-build-systems) more carefully now I'm assuming the correct argument to pass to `swiftc` is `-Xcc -std=c++20`? (or alternatively I was using `-std=gnu++11` with the project I had originally written in C++ and want to try and convert to swift, but I guess newer versions are generally better)

However, using either of these flags now gives the following error which I have no clue where to start with debugging - does `swiftc` get run twice, once for `C` and once for C++?:
```
$ idf.py build
...
/usr/bin/swiftc -j 8 -num-threads 8 -c ... -I ... ...
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'IDF_VER="v5.1.4"')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'MBEDTLS_CONFIG_FILE="mbedtls/esp_config.h"')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'SOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'ARDUINO=10812')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'ARDUINO_BOARD="ESP32C3_DEV"')
warning: conditional compilation flags do not have values in Swift; they are either present or absent (rather than 'ARDUINO_VARIANT="esp32c3"')
error: generate-pch command failed with exit code 1 (use -v to see invocation)
<unknown>:0: error: invalid argument '-std=c++20' not allowed with 'C'
<unknown>:0: error: clang importer creation failed

ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in build/log/idf_py_stderr_output_370636 build/log/idf_py_stdout_output_370636
```

maartin0, 2024-08-17:
Any ideas?

maartin0, 2024-08-28:
I think this just is a question of getting C++ to compile:

I'm not 100% clear on these [docs](https://www.swift.org/documentation/cxx-interop/project-build-setup/#mixing-swift-and-c-using-other-build-systems) but after re-reading I think `-cxx-interoperability-mode=default` can generate a clang module map if it doesn't find one. Re-including it gives the `undefined reference to \`_swift_exceptionPersonality\`` again which I have no clue where that's coming from, but further down in the docs it explains how you can use `-emit-clang-header-path SwiftModule-Swift.h` to emit the swift module if you use swift code in C++ sources. I'm not, but including it anyway gives a different error which I don't know if this is helpful, but seems to be slightly later in the process?

`swiftc` call:
```bash
/usr/bin/swiftc
-j 8 -num-threads 8 
-c 
-DESP_PLATFORM -DIDF_VER=\"v5.1.4\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS 
-parse-as-library -static -emit-module -emit-module-path esp-idf/main/__idf_main.swiftmodule -module-name __idf_main -module-link-name main 
-wmo -color-diagnostics -target riscv32-none-none-eabi 
-cxx-interoperability-mode=default -emit-clang-header-path SwiftModule-Swift.h 
-Xfrontend -function-sections -enable-experimental-feature Embedded 
-wmo -parse-as-library -Osize -Xcc -march=rv32imc_zicsr_zifencei -Xcc -mabi=ilp32 -Xcc -std=c++20 -Xcc -DARDUINO=10812 
-pch-output-dir /tmp
-Xfrontend -enable-single-module-llvm-emission 
-Xcc -I... 
-import-bridging-header main/BridgingHeader.h 
-v 
-DARDUINO=10812 -DARDUINO_ESP32C3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32C3_DEV\" -DARDUINO_VARIANT=\"esp32c3\" -DESP32 
-output-file-map esp-idf/main/CMakeFiles/__idf_main.dir//output-file-map.json 
-I build/config -I main 
-I ...
main/Main.swift main/Led.swift
```

Which calls `swift-frontend`, and gives the following error:
```bash
error: compile command failed due to signal 6 (use -v to see invocation)
swift-frontend: /home/build-user/swift/include/swift/IRGen/Linking.h:852: static swift::irgen::LinkEntity swift::irgen::LinkEntity::forTypeMetadata(swift::CanType, swift::irgen::TypeMetadataAddress): Assertion `!isEmbedded(concreteType) || isMetadataAllowedInEmbedded(concreteType)' failed.
Please submit a bug report (https://swift.org/contributing/#reporting-bugs) and include the crash backtrace.
Stack dump:
0.      Program arguments:
/usr/bin/swift-frontend -frontend
-c main/Main.swift main/Led.swift 
-emit-module-path esp-idf/main/__idf_main.swiftmodule 
-emit-module-doc-path esp-idf/main/__idf_main.swiftdoc 
-emit-module-source-info-path esp-idf/main/__idf_main.swiftsourceinfo 
-emit-objc-header-path SwiftModule-Swift.h 
-target riscv32-none-none-eabi 
-disable-objc-interop 
-cxx-interoperability-mode=default 
-I build/config -I main -I ~/esp-idf/...
-color-diagnostics 
-module-link-name main
-static
-Osize 
-D ESP_PLATFORM -D IDF_VER=\"v5.1.4\" -D MBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -D SOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D _GNU_SOURCE -D _POSIX_READER_WRITER_LOCKS -D ARDUINO=10812 -D ARDUINO_ESP32C3_DEV -D ARDUINO_ARCH_ESP32 -D ARDUINO_BOARD=\"ESP32C3_DEV\" -D ARDUINO_VARIANT=\"esp32c3\" -D ESP32
-function-sections 
-enable-single-module-llvm-emission
-enable-experimental-feature Embedded 
-empty-abi-descriptor 
-resource-dir /usr/lib/swift 
-Xcc -march=rv32imc_zicsr_zifencei -Xcc -mabi=ilp32 -Xcc -std=c++20 -Xcc -DARDUINO=10812 
-Xcc -I~/.espressif/...
-import-objc-header main/BridgingHeader.h
-pch-output-dir /tmp 
-module-name __idf_main 
-plugin-path /usr/lib/swift/host/plugins -plugin-path /usr/local/lib/swift/host/plugins
-enable-default-cmo -parse-as-library -num-threads 8 
-o esp-idf/main/CMakeFiles/__idf_main.dir/Main.swift.obj 
-o esp-idf/main/CMakeFiles/__idf_main.dir/Led.swift.obj
1.      Swift version 6.0-dev (LLVM 3602748d4ec9947, Swift 09b8b0861a69528)
2.      Compiling with effective version 5.10
3.      While generating Clang header
 #0 0x0000559b51ff30b7 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/usr/bin/swift-frontend+0x7aeb0b7)
 #1 0x0000559b51ff0e0e llvm::sys::RunSignalHandlers() (/usr/bin/swift-frontend+0x7ae8e0e)
 #2 0x0000559b51ff372a SignalHandler(int) Signals.cpp:0:0
 #3 0x00007fb0a2849320 (/lib/x86_64-linux-gnu/libc.so.6+0x45320)
[... 15 lines cut ...]
#19 0x0000559b4b4a2081 swift::ClangSyntaxPrinter::printNamespace(llvm::StringRef, llvm::function_ref<void (llvm::raw_ostream&)>, swift::ClangSyntaxPrinter::NamespaceTrivia) const (/usr/bin/swift-frontend+0xf9a081)
#20 0x0000559b4b4befec swift::printSwiftToClangCoreScaffold(swift::SwiftToClangInteropContext&, swift::ASTContext&, swift::PrimitiveTypeMapping&, llvm::raw_ostream&) (/usr/bin/swift-frontend+0xfb6fec)
#21 0x0000559b4b4aa8d6 swift::printModuleContentsAsCxx(llvm::raw_ostream&, swift::ModuleDecl&, swift::SwiftToClangInteropContext&, bool, llvm::StringSet<llvm::MallocAllocator>&) (/usr/bin/swift-frontend+0xfa28d6)
#22 0x0000559b4b498e22 void llvm::function_ref<void ()>::callback_fn<swift::printAsClangHeader(llvm::raw_ostream&, swift::ModuleDecl*, llvm::StringRef, swift::FrontendOptions const&, swift::IRGenOptions const&, clang::HeaderSearch&)::$_3>(long) PrintAsClang.cpp:0:0
#23 0x0000559b4b492cd7 emitCxxConditional(llvm::raw_ostream&, llvm::function_ref<void ()>, llvm::function_ref<void ()>) PrintAsClang.cpp:0:0
[... cut ...]
#36 0x00007fb0a282e28b call_init ./csu/../csu/libc-start.c:128:20
#37 0x00007fb0a282e28b __libc_start_main ./csu/../csu/libc-start.c:347:5
#38 0x0000559b4afcd3f5 _start (/usr/bin/swift-frontend+0xac53f5)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the build/log/idf_py_stderr_output_89909 and build/log/idf_py_stdout_output_89909
```

([full error](https://gist.githubusercontent.com/maartin0/5ce8f8f30acee2980b33706fb77e85d1/raw/1d1d80901c7d587b36bcf7bd8defbf34fd0929c2/gistfile1.txt))

maartin0, 2024-08-30:
I just noticed that `2.      Compiling with effective version 5.10` isn't the same as `6.0-dev` - could this be the cause of the missing symbols? Although with `-emit-clang-header-path SwiftModule-Swift.h` enabled, you get `Assertion '!isEmbedded(concreteType) || isMetadataAllowedInEmbedded(concreteType)' failed.` not the `undefined reference 'to _swift_exceptionPersonality'`

rauhul, 2024-09-26:
Update here: https://forums.swift.org/t/unable-to-build-image-using-embedded-swift-with-wifi-c-compiler-being-invoked/74934/5

maartin0, 2025-03-14:
For the `int16_t WiFiScanClass#scanNetworks(bool async = false, bool show_hidden = false, bool passive = false, uint32_t max_ms_per_chan = 300, uint8_t channel = 0, const char *ssid = nullptr, const uint8_t *bssid = nullptr)` error:
```cpp
undefined reference to `WiFiScanClass::scanNetworks(bool, bool, bool, unsigned int, unsigned char, char const*, unsigned char const*)
```
`libmain.a/Main.swift.obj` contains a reference to:
```
00000000         *UND*  00000000 WiFiScanClass::scanNetworks(bool, bool, bool, unsigned int, unsigned char, char const*, unsigned char const*)
```
Whereas the compiled `libespressif__arduino-esp32.a/WiFiScan.cpp.obj` has:
```
00000000 g     F .text._ZN13WiFiScanClass12scanNetworksEbbbmhPKcPKh     00000114 WiFiScanClass::scanNetworks(bool, bool, bool, unsigned long, unsigned char, char const*, unsigned char const*)
```

For some reason even though the signature has an `unsigned int` for argument 4 the compiled symbol table generated by `g++` is marked as an `unsigned long` - is there a missing flag to `swiftc` here?

[MWE](https://github.com/maartin0/swift-espidf-example) here

maartin0, 2025-08-13:
I've got a limited version of this working here [maartin0/arduino-swift-esp32-examples](https://github.com/maartin0/arduino-swift-esp32-examples)