swiftlang/swift-embedded-examples #45: Add example that builds with Xcode

rauhul, 2024-08-06:
We should have an example project that builds with xcodeproj for users in that ecosystem.

mredig, 2024-08-28:
This would make my workflow *much* easier if I knew how to do this.

PaulSolt, 2024-09-05:
What needs to be done to allow building in Xcode?

Are there a rough series of steps or configurations?

mredig, 2024-09-05:
Oh that's a good question - unless Paul wants to do this, I expect I'd be able to set it up given a little guidance.

rauhul, 2024-09-07:
So I think we should probably choose a simple project in this repo that could host examples for multiple build systems. I would like that example to consist of at least 1 module and an end binary using that module.

Actually building the example in Xcode should be pretty straightforward. We just need to setup an xcconfig file with some EXTRA_SWIFT_FLAGS and enable WMO as well as create an xcproj with 2 targets (library and executable)

mredig, 2024-09-08:
I've honestly never done anything outside the box like this with Xcode before. The most advanced skill I've gotten to myself is setting up some build configurations in the Xcode build settings. 

I've seen xcconfig files and others in other projects, but never really dug into why I'd use them over the built in build settings, so I only have a small bit of familiarity with this...

However the biggest blocker I have is how to manage the build environment for Espressif idf "stuff" (I don't know what to call it). I installed a supporting pip package in a venv (which I can't remember which one it was, just that it's a dep of idf.py) and, once I'm in that env, I need to activate the idf env with `export.sh`. I don't know how to go about leveraging these sub environments in Xcode (and I also wouldn't want to install into the global namespace on my computer, nor require it on others via an example)

ericlewis, 2025-03-31:
I took a crack at this with SPM and plugins, it’s a bit clunky.