swiftlang/swift-embedded-examples #48: SourceKit error: Cannot use metatype of type 'RCC.AHB1ENR.ReadWrite' in embedded Swift

mkbrwr, 2024-08-10:
While the stm32-uart-echo builds successfully, there are errors from SourceKit. Here's one of them 
[image]

kubamracek, 2024-08-10:
This is a false positive diagnostic coming from SourceKit, which is not present in regular builds, because regular builds have WMO (as required by Embedded Swift), but SourceKit sidesteps that and runs PerfDiags without WMO.

kubamracek, 2024-08-10:
I think this should address it: https://github.com/swiftlang/swift/pull/75308

closed 2024-08-13 by kubamracek

kubamracek, 2024-08-20:
Tried the latest downloadable toolchain and verified this is fixed now.