swiftlang/swift-embedded-examples #50: Build keeps failing

lawrencebensaid, 2024-08-18:
Hi all,

I'm a bit stuck. For a while I've been trying to get Swift to build for my ESP32 C6, but it keeps getting stuck in one of the last stages. I followed all the steps as described in [esp32-led-strip-sdk/README.md](esp32-led-strip-sdk/README.md).

Can someone point me in the right direction?

**Versions:**

```sh
$ idf.py --version
ESP-IDF v5.2.2-500-g2cd9da52fe
```
```sh
$ swift --version 
swift-driver version: 1.113 Apple Swift version 6.0 (swiftlang-6.0.0.7.6 clang-1600.0.24.1)
Target: arm64-apple-macosx15.0
```

**Build logs:**
```
➜  esp32-led-strip-sdk git:(main) idf.py build
...
[950/956] Building Swift Module '__idf_main' with 2 sources
FAILED: esp-idf/main/__idf_main.swiftmodule esp-idf/main/CMakeFiles/__idf_main.dir/Main.swift.obj esp-idf/main/CMakeFiles/__idf_main.dir/LedStrip.swift.obj 
/Applications/Xcode-beta.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/swiftc -j 10 -num-threads 10 -c -DESP_PLATFORM -DIDF_VER=\"v5.2.2-500-g2cd9da52fe\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUNITY_INCLUDE_CONFIG_H -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -parse-as-library -static -emit-module -emit-module-path esp-idf/main/__idf_main.swiftmodule -module-name __idf_main -module-link-name main -wmo -color-diagnostics -target riscv32-none-none-eabi -Xfrontend -function-sections -enable-experimental-feature Embedded -wmo -parse-as-library -Osize -Xcc -march=rv32imac_zicsr_zifencei -Xcc -mabi=ilp32 -pch-output-dir /tmp -Xfrontend -enable-single-module-llvm-emission -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/lib/gcc/riscv32-esp-elf/13.2.0/include -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/lib/gcc/riscv32-esp-elf/13.2.0/include-fixed -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/riscv32-esp-elf/sys-include -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/riscv32-esp-elf/include -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/riscv32-esp-elf/include/c++/13.2.0 -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/riscv32-esp-elf/include/c++/13.2.0/riscv32-esp-elf/rv32imac_zicsr_zifencei/ilp32 -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/riscv32-esp-elf/include/c++/13.2.0/backward -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/lib/gcc/riscv32-esp-elf/13.2.0/include -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/lib/gcc/riscv32-esp-elf/13.2.0/include-fixed -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/riscv32-esp-elf/sys-include -Xcc -I/Users/*****/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/riscv32-esp-elf/include -import-bridging-header /Users/*****/swift-embedded-examples/esp32-led-strip-sdk/main/BridgingHeader.h -output-file-map esp-idf/main/CMakeFiles/__idf_main.dir//output-file-map.json -I /Users/*****/swift-embedded-examples/esp32-led-strip-sdk/build/config -I /Users/*****/swift-embedded-examples/esp32-led-strip-sdk/main -I /Users/*****/esp/esp-idf/components/newlib/platform_include -I /Users/*****/esp/esp-idf/components/freertos/config/include -I /Users/*****/esp/esp-idf/components/freertos/config/include/freertos -I /Users/*****/esp/esp-idf/components/freertos/config/riscv/include -I /Users/*****/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I /Users/*****/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I /Users/*****/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I /Users/*****/esp/esp-idf/components/freertos/esp_additions/include -I /Users/*****/esp/esp-idf/components/esp_hw_support/include -I /Users/*****/esp/esp-idf/components/esp_hw_support/include/soc -I /Users/*****/esp/esp-idf/components/esp_hw_support/include/soc/esp32c6 -I /Users/*****/esp/esp-idf/components/esp_hw_support/port/esp32c6/. -I /Users/*****/esp/esp-idf/components/esp_hw_support/port/esp32c6/private_include -I /Users/*****/esp/esp-idf/components/heap/include -I /Users/*****/esp/esp-idf/components/log/include -I /Users/*****/esp/esp-idf/components/soc/include -I /Users/*****/esp/esp-idf/components/soc/esp32c6 -I /Users/*****/esp/esp-idf/components/soc/esp32c6/include -I /Users/*****/esp/esp-idf/components/hal/platform_port/include -I /Users/*****/esp/esp-idf/components/hal/esp32c6/include -I /Users/*****/esp/esp-idf/components/hal/include -I /Users/*****/esp/esp-idf/components/esp_rom/include -I /Users/*****/esp/esp-idf/components/esp_rom/include/esp32c6 -I /Users/*****/esp/esp-idf/components/esp_rom/esp32c6 -I /Users/*****/esp/esp-idf/components/esp_common/include -I /Users/*****/esp/esp-idf/components/esp_system/include -I /Users/*****/esp/esp-idf/components/esp_system/port/soc -I /Users/*****/esp/esp-idf/components/esp_system/port/include/riscv -I /Users/*****/esp/esp-idf/components/esp_system/port/include/private -I /Users/*****/esp/esp-idf/components/riscv/include -I /Users/*****/esp/esp-idf/components/lwip/include -I /Users/*****/esp/esp-idf/components/lwip/include/apps -I /Users/*****/esp/esp-idf/components/lwip/include/apps/sntp -I /Users/*****/esp/esp-idf/components/lwip/lwip/src/include -I /Users/*****/esp/esp-idf/components/lwip/port/include -I /Users/*****/esp/esp-idf/components/lwip/port/freertos/include -I /Users/*****/esp/esp-idf/components/lwip/port/esp32xx/include -I /Users/*****/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I /Users/*****/esp/esp-idf/components/lwip/port/esp32xx/include/sys -I /Users/*****/swift-embedded-examples/esp32-led-strip-sdk/managed_components/espressif__led_strip/include -I /Users/*****/swift-embedded-examples/esp32-led-strip-sdk/managed_components/espressif__led_strip/interface -I /Users/*****/esp/esp-idf/components/driver/include -I /Users/*****/esp/esp-idf/components/driver/deprecated -I /Users/*****/esp/esp-idf/components/driver/analog_comparator/include -I /Users/*****/esp/esp-idf/components/driver/dac/include -I /Users/*****/esp/esp-idf/components/driver/gpio/include -I /Users/*****/esp/esp-idf/components/driver/gptimer/include -I /Users/*****/esp/esp-idf/components/driver/i2c/include -I /Users/*****/esp/esp-idf/components/driver/i2s/include -I /Users/*****/esp/esp-idf/components/driver/ledc/include -I /Users/*****/esp/esp-idf/components/driver/mcpwm/include -I /Users/*****/esp/esp-idf/components/driver/parlio/include -I /Users/*****/esp/esp-idf/components/driver/pcnt/include -I /Users/*****/esp/esp-idf/components/driver/rmt/include -I /Users/*****/esp/esp-idf/components/driver/sdio_slave/include -I /Users/*****/esp/esp-idf/components/driver/sdmmc/include -I /Users/*****/esp/esp-idf/components/driver/sigma_delta/include -I /Users/*****/esp/esp-idf/components/driver/spi/include -I /Users/*****/esp/esp-idf/components/driver/temperature_sensor/include -I /Users/*****/esp/esp-idf/components/driver/touch_sensor/include -I /Users/*****/esp/esp-idf/components/driver/twai/include -I /Users/*****/esp/esp-idf/components/driver/uart/include -I /Users/*****/esp/esp-idf/components/driver/usb_serial_jtag/include -I /Users/*****/esp/esp-idf/components/esp_pm/include -I /Users/*****/esp/esp-idf/components/esp_ringbuf/include -I /Users/*****/esp/esp-idf/components/efuse/include -I /Users/*****/esp/esp-idf/components/efuse/esp32c6/include -I /Users/*****/esp/esp-idf/components/esp_mm/include -I /Users/*****/esp/esp-idf/components/mbedtls/port/include -I /Users/*****/esp/esp-idf/components/mbedtls/mbedtls/include -I /Users/*****/esp/esp-idf/components/mbedtls/mbedtls/library -I /Users/*****/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I /Users/*****/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I /Users/*****/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I /Users/*****/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I /Users/*****/esp/esp-idf/components/esp_bootloader_format/include -I /Users/*****/esp/esp-idf/components/esp_app_format/include -I /Users/*****/esp/esp-idf/components/bootloader_support/include -I /Users/*****/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I /Use [... cut ...]
21 swift-frontend           0x00000001051a850c swift::mainEntry(int, char const**) + 3680
22 dyld                     0x0000000190fc0274 start + 2840
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/*****/swift-embedded-examples/esp32-led-strip-sdk/build/log/idf_py_stderr_output_5601 and /Users/*****/swift-embedded-examples/esp32-led-strip-sdk/build/log/idf_py_stdout_output_5601
```

Any suggestions of help would be greatly appreciated 🙏🏼
Thanks!

kubamracek, 2024-08-19:
The error message includes:

```
/Applications/Xcode-beta.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/swiftc ...
```

which means you are using Swift from the Xcode installation, but that won't work, you need to use a downloaded toolchain from swift.org. Did you maybe forget to `export TOOLCHAINS=...` before running `idf.py set-target`?

rauhul, 2025-01-03:
Closing this as stale, I believe Kuba's diagnosis is correct and moving to a nightly toolchain will resolve this issue. We now have CI that covers building this example which can also be used as a reference.

closed 2025-01-03 by rauhul