swiftlang/swift-embedded-examples #55: Serial (UART?) input

mredig, 2024-08-22:
I've found that I'm able to `print` via Swift on my esp32c*, but I cannot figure out how to read serial input.

Is this built into Swift at all, or do I need to drop into c calls?

kubamracek, 2024-08-23:
Even UART output is not really built into Embedded Swift -- the standard Swift functions that produce text output like `print()` are currently just redirected to putchar() calls, and we expect that the underlaying platform provides that. IIUC, the ESP SDK routes putchar into the JTAG-forwarded UART (on JTAG enabled boards).

So I think the solution here is to manually operate a UART in your code, especially if you want to use a different UART thatn the one that's used for normal logging from the various subsystems in ESP SDK. So, probably you're looking at library calls to the ESP SDK, or alternatively accessing the UART HW directly similarly to how this STM32 example code is set up: https://github.com/apple/swift-embedded-examples/blob/main/stm32-uart-echo/Sources/Application/Application.swift

rauhul, 2025-05-18:
We could update the STM32 UART example to use `readLine` and implement `_swift_stdlib_readLine_stdin` over UART RX. Maybe that would be a good addition