swiftlang/swift-embedded-examples #56: String.count -> UInt8 link fail

mredig, 2024-08-23:
So far, the following sample code has failed in two separate projects for me:

```swift
let a = "asdf".count
let b = UInt8(a)
```

Compilation appears to succeed, but fail on the linking step. Let me know if you need more to go on.

Oh - working with espidf and compiling to an esp32c3, if any of that matters.

rauhul, 2024-10-11:
Is this still an issue with new toolchains? I was not able to reproduce any issues with the above snippet:
```
➜  Desktop cat foo.swift
let a = "asdf".count
let b = UInt8(a)
➜  Desktop swiftc foo.swift -emit-object -o foo.o -enable-experimental-feature Embedded -wmo -target armv7em-apple-none-macho -Xcc -D__APPLE__ -Xcc -D__MACH__
➜  Desktop swiftc --version
Apple Swift version 6.1-dev (LLVM 89ccf4b8a46135a, Swift 6a5ae8d5df144dd)
Target: arm64-apple-macosx15.0
```

rauhul, 2025-05-03:
Closing this one, please reopen if you are able to repro.

closed 2025-05-03 by rauhul