swiftlang/swift-embedded-examples #62: pico-w-blink-sdk | cmake --build build | fail

LunaTMT, 2024-10-09:
$ cd pico-w-blink-sdk

```
echo $TOOLCHAINS
echo $PICO_BOARD
echo $PICO_SDK_PATH
echo $PICO_TOOLCHAIN_PATH

swift-6.0.1
pico_w
/home/taylor/pico-sdk
/usr/bin/
   
which arm-none-eabi-gcc
/usr/bin/arm-none-eabi-gcc
```

First build with Ninja works fine:
```
$ cmake -B build -G Ninja .

PICO_SDK_PATH is /home/taylor/pico-sdk
Target board (PICO_BOARD) is 'pico_w'.
Using board configuration from /home/taylor/pico-sdk/src/boards/include/boards/pico_w.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
Build type is Release
TinyUSB available at /home/taylor/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/taylor/pico-sdk/lib/btstack
cyw43-driver available at /home/taylor/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/taylor/pico-sdk/lib/lwip
Pico W Wi-Fi build support available.
mbedtls available at /home/taylor/pico-sdk/lib/mbedtls
-- Configuring done (0.2s)
-- Generating done (0.0s)
-- Build files have been written to: /home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/b

```

Running cmake build:

```
$cmake --build build
[1/77] Generating _swiftcode.o
FAILED: _swiftcode.o /home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/build/_swiftcode.o 
cd /home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/build && /usr/local/swift/usr/bin/swiftc -target armv6m-none-none-eabi -Xcc -mfloat-abi=soft -Xcc -fshort-enums -Xcc -DCYW43_LWIP -Xcc -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND -Xcc -I~/pico-sdk/lib/lwip/src/include -Xcc -I/home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/include -Xcc -I~/pico-sdk/src/common/pico_stdlib_headers/include -Xcc -I~/pico-sdk/src/rp2040/hardware_structs/include -Xcc -I~/pico-sdk/src/rp2_common/hardware_base/include -Xfrontend -function-sections -enable-experimental-feature Embedded -wmo -parse-as-library $( echo '/home/taylor/pico-sdk/src/common/pico_stdlib_headers/include;/home/taylor/pico-sdk/src/rp2_common/hardware_gpio/include;/home/taylor/pico-sdk/src/common/pico_base_headers/include;/home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/build/generated/pico_base;/home/taylor/pico-sdk/src/boards/include;/home/taylor/pico-sdk/src/rp2040/pico_platform/include;/home/taylor/pico-sdk/src/rp2040/hardware_regs/include;/home/taylor/pico-sdk/src/rp2_common/hardware_base/include;/home/taylor/pico-sdk/src/rp2_common/pico_platform_compiler/include;/home/taylor/pico-sdk/src/rp2_common/pico_platform_panic/include;/home/taylor/pico-sdk/src/rp2_common/pico_platform_sections/include;/home/taylor/pico-sdk/src/rp2040/hardware_structs/include;/home/taylor/pico-sdk/src/common/hardware_claim/include;/home/taylor/pico-sdk/src/rp2_common/hardware_sync/include;/home/taylor/pico-sdk/src/rp2_common/hardware_sync_spin_lock/include;/home/taylor/pico-sdk/src/rp2_common/hardware_irq/include;/home/taylor/pico-sdk/src/rp2_common/hardware_uart/include;/home/taylor/pico-sdk/src/rp2_common/hardware_resets/include;/home/taylor/pico-sdk/src/rp2_common/hardware_clocks/include;/home/taylor/pico-sdk/src/rp2_common/hardware_pll/include;/home/taylor/pico-sdk/src/rp2_common/hardware_vreg/include;/home/taylor/pico-sdk/src/rp2_common/hardware_watchdog/include;/home/taylor/pico-sdk/src/rp2_common/hardware_ticks/include;/home/taylor/pico-sdk/src/rp2_common/hardware_xosc/include;/home/taylor/pico-sdk/src/rp2_common/hardware_divider/include;/home/taylor/pico-sdk/src/common/pico_time/include;/home/taylor/pico-sdk/src/rp2_common/hardware_timer/include;/home/taylor/pico-sdk/src/common/pico_sync/include;/home/taylor/pico-sdk/src/common/pico_util/include;/home/taylor/pico-sdk/src/rp2_common/pico_time_adapter/include;/home/taylor/pico-sdk/src/rp2_common/pico_runtime/include;/home/taylor/pico-sdk/src/rp2_common/pico_runtime_init/include;/home/taylor/pico-sdk/src/common/pico_bit_ops_headers/include;/home/taylor/pico-sdk/src/common/pico_divider_headers/include;/home/taylor/pico-sdk/src/rp2_common/pico_double/include;/home/taylor/pico-sdk/src/rp2_common/pico_float/include;/home/taylor/pico-sdk/src/rp2_common/pico_malloc/include;/home/taylor/pico-sdk/src/rp2_common/pico_atomic/include;/home/taylor/pico-sdk/src/common/pico_binary_info/include;/home/taylor/pico-sdk/src/rp2_common/pico_printf/include;/home/taylor/pico-sdk/src/rp2_common/pico_stdio/include;/home/taylor/pico-sdk/src/rp2_common/pico_stdio_uart/include;/home/taylor/pico-sdk/src/rp2_common/pico_bootrom/include;/home/taylor/pico-sdk/src/common/boot_picoboot_headers/include;/home/taylor/pico-sdk/src/rp2_common/hardware_boot_lock/include;/home/taylor/pico-sdk/src/rp2_common/pico_int64_ops/include;/home/taylor/pico-sdk/src/rp2_common/pico_mem_ops/include;/home/taylor/pico-sdk/src/rp2040/boot_stage2/include;/home/taylor/pico-sdk/src/common/boot_picobin_headers/include;/home/taylor/pico-sdk/src/rp2_common/pico_lwip/include;/home/taylor/pico-sdk/src/rp2_common/pico_rand/include;/home/taylor/pico-sdk/src/rp2_common/pico_unique_id/include;/home/taylor/pico-sdk/src/rp2_common/hardware_flash/include;/home/taylor/pico-sdk/src/rp2_common/pico_cyw43_arch/include;/home/taylor/pico-sdk/lib/cyw43-driver/src;/home/taylor/pico-sdk/lib/cyw43-driver/firmware;/home/taylor/pico-sdk/src/rp2_common/pico_cyw43_driver/cybt_shared_bus;/home/taylor/pico-sdk/src/rp2_common/hardware_pio/include;/home/taylor/pico-sdk/src/rp2_common/hardware_dma/include;/home/taylor/pico-sdk/src/rp2_common/hardware_exception/include;/home/taylor/pico-sdk/src/rp2_common/pico_cyw43_driver/include;/home/taylor/pico-sdk/src/rp2_common/pico_async_context/include;/home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/build/pico-sdk/src/rp2_common/pico_cyw43_driver' | tr ';' '\n' | sed -e 's/\(.*\)/-Xcc -I\1/g' ) $( echo '/usr/lib/gcc/arm-none-eabi/13.2.1/include /usr/lib/gcc/arm-none-eabi/13.2.1/include-fixed /usr/lib/arm-none-eabi/include' | tr ' ' '\n' | sed -e 's/\(.*\)/-Xcc -I\1/g' ) -import-bridging-header /home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/BridgingHeader.h /home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/Main.swift -c -o /home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/build/_swiftcode.o
error: generate-pch command failed with exit code 1 (use -v to see invocation)
/home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/BridgingHeader.h:14:10: note: in file included from /home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/BridgingHeader.h:14:
12 | #pragma once
13 | 
14 | #include "pico/stdlib.h"
   |          `- note: in file included from /home/taylor/Desktop/swift-embedded-examples/pico-w-blink-sdk/BridgingHeader.h:14:
15 | #include "pico/cyw43_arch.h"
16 | 

/home/taylor/pico-sdk/src/common/pico_stdlib_headers/include/pico/stdlib.h:12:10: note: in file included from /home/taylor/pico-sdk/src/common/pico_stdlib_headers/include/pico/stdlib.h:12:
10 | #include "pico.h"
11 | #include "pico/stdio.h"
12 | #include "pico/time.h"
   |          `- note: in file included from /home/taylor/pico-sdk/src/common/pico_stdlib_headers/include/pico/stdlib.h:12:
13 | #include "hardware/gpio.h"
14 | #include "hardware/uart.h"

/home/taylor/pico-sdk/src/common/pico_time/include/pico/time.h:11:10: note: in file included from /home/taylor/pico-sdk/src/common/pico_time/include/pico/time.h:11:
  9 | 
 10 | #include "pico.h"
 11 | #include "hardware/timer.h"
    |          `- note: in file included from /home/taylor/pico-sdk/src/common/pico_time/include/pico/time.h:11:
 12 | 
 13 | #ifdef __cplusplus
[... 20 lines cut ...]
175 | #if !PICO_RP2040
176 | // include this here to avoid the check in every other hardware/structs header that needs it
177 | #include "hardware/structs/accessctrl.h"
    |          `- error: 'hardware/structs/accessctrl.h' file not found
178 | #endif
179 | 
ninja: build stopped: subcommand failed.
```

On checking the existence of "accessctrl.h" it cannot be found 
```
~/pico-sdk/src/rp2040/hardware_structs/include/hardware/structs/accessctrl.h
bash: /home/taylor/pico-
```sdk/src/rp2040/hardware_structs/include/hardware/structs/accessctrl.h: No such file or directory

But the pico-sdk is already up to date on master branch
```
$ cd ~/pico-sdk
git pull
Already up to date.

~/pico-sdk$ git branch
* master

```

iCMDdev, 2024-10-10:
Hi! This is actually described in #59. It's a pretty long thread but it has a (temporary) solution I found. I'll make sure to create a PR in the near future.

Long story short, why this happens:  the Pico SDK compiler definitions are not passed to `swiftc`, so the compiler somehow thinks you're building for RP2350.

rdlaner, 2024-10-20:
@iCMDdev , I'm just getting started with embedded Swift and thanks to your info here was able to get the blinky example working on the pico-w. However, even after applying your changes I was still getting errors regarding missing cyw header files. As a bit of a hack, I got it working by adding these libraries, `pico_lwip_arch pico_cyw43_arch_none`, back to the `target_link_libraries` call as well as some of the CYW43 definitions. Looks like this:

```
target_link_libraries(swift-blinky
    pico_stdlib hardware_uart hardware_gpio pico_lwip_arch pico_cyw43_arch_none
)
```

```
add_custom_command(
    OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
    COMMAND
        ${SWIFTC}
        -target ${SWIFT_TARGET} -Xcc -fshort-enums
        @${CMAKE_BINARY_DIR}/swiftc_flags.txt
        ${CLANG_ARCH_ABI_FLAGS}
        -Xcc -DCYW43_LWIP
        -Xcc -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND
        -Xcc -I$ENV{PICO_SDK_PATH}/lib/lwip/src/include
        -Xcc -I${CMAKE_CURRENT_LIST_DIR}/include
        -Xfrontend -function-sections -enable-experimental-feature Embedded -wmo -parse-as-library
        $$\( echo '$<TARGET_PROPERTY:swift-blinky,INCLUDE_DIRECTORIES>' | tr '\;' '\\n' | sed -e 's/\\\(.*\\\)/-Xcc -I\\1/g' \)
        $$\( echo '${CMAKE_C_IMPLICIT_INCLUDE_DIRECTORIES}'             | tr ' '  '\\n' | sed -e 's/\\\(.*\\\)/-Xcc -I\\1/g' \)
        -import-bridging-header ${CMAKE_CURRENT_LIST_DIR}/BridgingHeader.h
        ${CMAKE_CURRENT_LIST_DIR}/Main.swift
        -c -o ${CMAKE_CURRENT_BINARY_DIR}/_swiftcode.o
    DEPENDS
        ${CMAKE_CURRENT_LIST_DIR}/BridgingHeader.h
        ${CMAKE_CURRENT_LIST_DIR}/Main.swift
)
```

I can't give much of an explanation yet, but I wondered if you encountered anything like this if you were building for the pico-w?

iCMDdev, 2024-10-20:
@rdlaner Pretty fun coincidence that I checked my email right when you replied 😆

> However, even after applying your changes I was still getting errors regarding missing cyw header files. As a bit of a hack, I got it working by adding these libraries, pico_lwip_arch pico_cyw43_arch_none, back to the target_link_libraries call as well as some of the CYW43 definitions. 

Right, yes, you need to include those headers as well for the Pico W. I wouldn't say this is a hacky way to do it, I think it's normal / expected. My `CMakeLists.txt` example was pretty general one, made to work.

> but I wondered if you encountered anything like this if you were building for the pico-w?

Yeah, I also had to include those headers, and I'd say this is normal / expected. Check out [the Pico W example in the PR I recently made](https://github.com/apple/swift-embedded-examples/blob/e6932490c847cce4b0705fa219c1ae7877aa9a36/pico-w-blink-sdk/CMakeLists.txt).

iCMDdev, 2024-11-13:
Hey @LunaTMT, is it working for you now? PR was merged. I think this can be closed.

rauhul, 2025-03-20:
Given that we have CI covering this build im pretty confident we can close this issue. feel free to reopen it if needed

closed 2025-03-20 by rauhul