swiftlang/swift-embedded-examples #65: Request to add repository to Community Examples list for Embedded Swift

ebariaux, 2024-10-23:
Hello,

I've been writing a few simple examples of using Embedded Swift on nRF52840dk using the nRF Connect SDK.
They are available at [nelcea/EmbeddedSwift-nRF52-Examples: Examples of using Embedded Swift to program nRF52xxx microcontrollers](https://github.com/nelcea/EmbeddedSwift-nRF52-Examples).
There are also blog posts accompanying them at https://www.ericbariaux.com
And I intend to keep adding posts / code examples as time allows and I broaden my use of Embedded Swift.

Name:  [EmbeddedSwift nRF52 Examples](https://github.com/nelcea/EmbeddedSwift-nRF52-Examples)
Platform: Nordic Semiconductor nRF52840dk
Description: Collection of examples using Embedded Swift on top of nRF Connect SDK (Zephyr)

Thank you for considering this addition,
Best regards,
Eric

rauhul, 2024-12-15:
We'd be happy to include this link, please open a PR with a link

ebariaux, 2024-12-20:
Thank you, see [Add 2 community examples running on nRF52840 microcontroller](https://github.com/apple/swift-embedded-examples/pull/82)

closed 2025-01-03 by rauhul