swiftlang/swift-embedded-examples #92: Include Unit Testing examples

tdcox, 2025-03-11:
Ideally, educational examples should include some simple unit tests to demonstrate the requisite patterns for testing in a new environment.

This is particularly important for embedded systems since cross-compilation introduces significant testing challenges. There are several scenarios that should be demonstrated:

1. Testing Swift 'business logic' on the host machine - compiling for the host and executing tests locally as part of the build.
2. Testing on target hardware - cross-compiling and then launching a test suite on target hardware and collecting results.
3. Testing on emulated hardware - cross-compiling and then launching a test suite on a simulator / emulator running on the host.

Consideration should be given to managing external dependencies like FreeRTOS libraries and physical hardware that may have limited emulation support.

rauhul, 2025-05-03:
This is a great idea :) Do you have an idea of what build systems you'd be interested in seeing this with? For 2 / 3 I have had success with bazel + embedded swift.

tdcox, 2025-05-03:
A good example is the ESP32 ecosystem. This has to sit on top of the ESP-IDF tool chain to build, which requires deep magic to mess with.

In Rust, you would create a main crate that was dependent upon the Rust esp-idf-svc library and aware of how to target the riscv platform without specialist knowledge. You could then create independent pure Rust libraries for business logic and include unit tests within them. These could then be imported into the main crate as required.

These can then be wrapped in a cargo workspace (a parent folder with some meta-data) that includes a build script for any CI/CD system.

If the script executes 'cargo test' in each of the pure Rust libraries in turn, that library codebase would be built for the build platform and tested locally on the fly (Scenario 1).

If the script then builds in the main crate, a binary for the riscv target would be built.

This can be executed under the Wokwi simulator (Scenario 3) and then deployed to a physical device as-is, or built by injecting a remote unit testing framework and then unit tested on the physical device (Scenario 2) before rebuilding as a test-free production binary.

In the Rust eco-system, all this comes pretty much out-of-the-box with a handful of commands to create the projects from templates, including a basic build script for GitHub Actions. This makes it very easy for novice users to start from a known good development environment and start exploring the features of the language and environment through unit testing.

The lack of a similar ease of use is a big barrier to adoption for embedded swift since the main argument for adoption is the ability to write higher quality code, but this falls down if you can't test and integrate into your Continuous Delivery pipeline.