swiftlang/swift-embedded-examples #102: Can't build basic examples doing step-by-step from guide

truebest, 2025-03-31:
Hi folks,

Once I failed with ubuntu docker, taken my MacBook Apple Silicon and tried to build default examples based on apple example [docs](https://apple.github.io/swift-matter-examples/)  

Current swift snapshot: swift-DEVELOPMENT-SNAPSHOT-2025-03-28-a-osx.pkg

during building led-blink " idf.py build " got follow error

```
[710/1257] Performing configure step for 'chip_gn'
Done. Made 253 targets from 211 files in 115ms
[711/1257] Performing build step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /Users/truebest/swift-matter-examples/led-blink/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /Users/truebest/swift-matter-examples/led-blink/build/esp-idf/chip/lib/libCHIP.a 
cd "/Users/truebest/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;/Users/truebest/swift-matter-examples/led-blink/build/esp-idf/chip" && ninja esp32
/bin/sh: line 0: cd: /Users/truebest/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;/Users/truebest/swift-matter-examples/led-blink/build/esp-idf/chip: No such file or directory
[713/1257] Performing configure step for 'bootloader'
```

also I tried to build other example, got around same issue.

```
`FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /Users/truebest/swift-matter-examples/smart-light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /Users/truebest/swift-matter-examples/smart-light/build/esp-idf/chip/lib/libCHIP.a 
cd "/Users/truebest/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;/Users/truebest/swift-matter-examples/smart-light/build/esp-idf/chip" && ninja esp32
/bin/sh: line 0: cd: /Users/truebest/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;/Users/truebest/swift-matter-examples/smart-light/build/esp-idf/chip: No such file or directory
`
```

iCMDdev, 2025-04-22:
Seems related to the Espressif SDK, did you install it properly? Does ESP-IDF work fine for you?

Also, based on the paths, I think you're referring to the [swift-matter-examples](https://github.com/apple/swift-matter-examples) repo.

kubamracek, 2025-04-26:
Did you try the Espressif SDK (IDF) + the Matter SDK (ESP Matter) without Swift? This command that's generated by those SDKs seems completely wrong (it's trying to cd into something that has a ";" in the middle):

```
cd "/Users/truebest/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip;/Users/truebest/swift-matter-examples/led-blink/build/esp-idf/chip" && ninja esp32
```