swiftlang/swift-embedded-examples #139: C++ Interop Example

ADKaster, 2025-05-19:
I took a crack at trying to create an example project for including a Swift library as a dependency to an existing C++ Zephyr project. Well by "existing" I mean, the app main is a C++ file, not that I have an existing Zephyr project to integrate with.

That example lives here: https://github.com/ADKaster/swift-embedded-export-zephyr

I based it off of some of the examples in this repository for Zephyr.

It's possible that "Zephyr + C++" is too tricky of a target to create this interop example for embedded from. But I doubt that this is a problem unique to me.

Would an example in a different operating environment be a good fit for this repo?

rauhul, 2025-05-23:
Yeah this would be a good thing to include in an example attached to the integrate with zephyr guide in #140

xtremekforever, 2025-05-24:
This seems interesting, @ADKaster do you want a crack at adding the example to the document or shall I have a look later?

ADKaster, 2025-05-24:
Also iirc I filed https://github.com/swiftlang/swift/issues/80570 while playing with this example. Evan's CMake scripts attempt to extract information about target-specific include dirs and lib dirs from `swiftc -print-target-info`. But this completely falls over when using `-enable-experimental-feature Embedded`. 

Relevant script lines:
https://github.com/ADKaster/swift-embedded-export-zephyr/blob/main/cmake/InitializeSwift.cmake#L39